ARM-software/LLVM-embedded-toolchain-for-Arm #81: Support Zephyr Project as an ARM official toolchain

andy0808, 2021-11-24:
[Zephyr Project](https://github.com/zephyrproject-rtos/zephyr) is becoming more and more popular in the RTOS area. For ARM Cortex architecture, they have their own toolchain/SDK based on GCC. And the ARM maintained arm-none-eabi-gcc also works perfectly with this project.

They have discussed [supporting LLVM a few years ago](https://github.com/zephyrproject-rtos/zephyr/issues/3689), but seems there's not any progress yet! I think the main reason is that there's no official LLVM toolchain maintained for ARM Cortex before.

Since you are supporting LLVM officially, is there any schedule to support these modern RTOS? Thank you : )

I think combining LLVM + modern RTOS makes it easier to use modern languages on microcontrollers such as Rust and Swift, they both are based on LLVM : ) This would help application development on microcontrollers a lot!

lindblandro, 2022-01-23:
I think you could compile Rust code as a library by suitably declaring the exported interface (no name mangling and extern "C") and include that to your application using `ExternalProject`, maybe add some scripts to export C headers for the library for convenience. Having a binary crate as the cross-compiled application might be a lot trickier, but nice. There's also https://github.com/SiegeLord/RustCMake which doesn't seem to be maintained, but might serve as a starting point for something.

adityashah1212, 2023-07-21:
@lindblandro, I think that is one way to do it, but brings in one issue. cross-language LTO won't be possible. I myself found LTO can save space quite a bit in the final binary and can also improve speeds quite a bit. Having LLVM directly support will improve on that.

voltur01, 2023-07-24:
I noticed this issue https://github.com/zephyrproject-rtos/zephyr/issues/54826 in Zephyr project that refers to LLVMEmbeddedToolchainForArm-15.0.2, however seems to be stale by now.

voltur01, 2025-01-06:
This seems to be well in progress with https://github.com/zephyrproject-rtos/sdk-ng/issues/777 so I will close this issue as a duplicate.

closed 2025-01-06 by voltur01