ARM-software/LLVM-embedded-toolchain-for-Arm #83: Windows version of the toolchain, easy to freeze when linking

Guozhanxin, 2021-12-22:
I have supported the LLVM for Arm toolchain on the RT-Thread open source operating system, but I found a problem during use.

When I use it in a window environment, after the compilation is complete, there is a high probability of being stuck in the step of linking to generate elf. There is no such problem under WSL(ubuntu 20.04).

https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32l496-ali-developer

### windows
[image]
### WSL(ubuntu 20.04)
[image]

voltur01, 2021-12-22:
Thank you for using the toolchain and reporting the issue!

Would you be able to provide a bit more diagnostic, e.g. by attaching a debugger or using a utility like https://docs.microsoft.com/en-us/sysinternals/downloads/procdump - something like stack trace that could help figure out if the problem is in the linker or the mingw runtime.

I assume that "high probability of being stuck" means that sometimes the build completes successfully.

Guozhanxin, 2021-12-23:
[clang.exe_211223_103700.zip](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/files/7766300/clang.exe_211223_103700.zip)

voltur01, 2021-12-23:
Thank you! We shall have a look and update the issue accordingly.

SquallATF, 2021-12-26:
may be should cross build llvm use clang like https://github.com/mstorsjo/llvm-mingw

voltur01, 2022-01-04:
Thank you for the link!

Guozhanxin, 2022-01-15:
Do you have any plans to fix this issue?

voltur01, 2022-01-17:
Sure, we shall review and decide on next steps (likely the issue is in one of upstream projects), however I cannot give a specific due date yet. We shall update as soon as there are any more details.

astrolabe11, 2022-04-27:
By the comment above, I gave it a try building on ubuntu 20.0.4 with the options here, it looks fine so far.

$ build.py --use-ninja --host-toolchain mingw --native-toolchain clang

amilendra, 2022-07-29:
@Guozhanxin 
Can you provide more information on how to reproduce this please (command lines used to build the project etc)? If possible also check if this still appears with the latest release (https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases/tag/release-14.0.0)

Guozhanxin, 2022-08-01:
> @Guozhanxin Can you provide more information on how to reproduce this please (command lines used to build the project etc)? If possible also check if this still appears with the latest release (https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases/tag/release-14.0.0)

Great! I will test this new version when I have time in the near future.

closed 2023-04-17 by mplatings

Guozhanxin, 2023-05-19:
Great job, now it's working great!  https://github.com/RT-Thread/rt-thread/pull/7521