ARM-software/LLVM-embedded-toolchain-for-Arm #84: Missing libc_nano

vsydoruk, 2021-12-29:
Hello,

First of all , it is great to see LLVM based embedded toolchain is coming for ARM Cortex M family. Thank you guys for the effort.

Now to the issue, contrary to the ARM GNU Embedded toolchain there is no "nano" flavored C library in the output directory and the released binary. There is, however, the nano.specs file present which is a bit odd.  Is there a plan to compile newlib with "nano" configuration? 

Thank you,
Volod

voltur01, 2022-01-04:
Hi Volod,

Thanks for checking out the toolchain!

The nano.specs file, as well as other spec files in the lib directory, is standard output from newlib build that we did not remove specifically even though the toolchain is not using it - the configuration is provided in the *.cfg files found in the bin directory.

We do expect to add more library options in the future including a minimal version like nano for really small MCUs, however there is no particular due date for that yet.

I hope this answers the question.

vsydoruk, 2022-01-04:
Thank you for the answer.
I'd like to experiment with the build system and get the libc_nano equivalent to the ARM GCC toolchain compiled for testing.
Do you know what are the newlib configuration options used for nano build for  ARM GNU Embedded toolchain?

voltur01, 2022-01-04:
This project https://github.com/ARM-software/toolchain-gnu-bare-metal may be helpful, see `build-toolchain.sh`

GorgonMeducer, 2022-05-03:
It looks like the latest LLVM 14 release still doesn't provide the nano c library.

voltur01, 2022-05-03:
Hi Gabriel,

Yes, we want to sort out the multilib support https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues/110 before adding more library options, so for now only regular newlib is included.

voltur01, 2022-08-02:
The library was switched to picolibc as per https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues/61, so I shall close this issue, at least for now.

Here is the comparison of picolibc vs newlib-nano https://github.com/picolibc/picolibc/issues/103

closed 2022-08-02 by voltur01