ARM-software/LLVM-embedded-toolchain-for-Arm #86: can release osx binary？

touchmii, 2022-01-06:


voltur01, 2022-01-06:
Hi, we focus on Linux and Windows support for now, macOS may come later. 

Note that some users reported they were able to successfully build on macOS themselves, e.g. see https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/62

flit, 2022-02-22:
Not supporting macOS support is pretty unacceptable these days. Many software/embedded developers use Macs because doing so allows easy management of all three major OSes through VMs and Docker. And there are probably more internal Mac users at Arm than any other OS.

voltur01, 2022-02-23:
Agreed, macOS support is on the roadmap, only a question of priority versus other tasks. We shall do our best not to postpone for too long.

nimish, 2022-05-10:
Wouldn't this just need an extra step with `-target={x86_64,aarch64}-apple-darwin` on the build for release pipeline? Building llvm manually is a chore. 

The lack of support for `aarch64-apple-darwin` is quite ironic.

stuij, 2023-02-24:
I made a temporary patch to make this work for OSX if you want to go through the trouble of building from source:
https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/156

mplatings, 2023-04-17:
Now available: https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases/download/release-16.0.0/LLVMEmbeddedToolchainForArm-16.0.0-Darwin.tar.gz

Note that the way it's packaged doesn't play nicely with Gatekeeper so you'll need to work around that: https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm#getting-started. We intend to change the packaging in a future release.

closed 2023-04-17 by mplatings