ARM-software/LLVM-embedded-toolchain-for-Arm #87: Error when compiling micro-ecc

zxtxin, 2022-01-17:
I build micro-ecc with the following parameters:
`-std=c11 --config armv6m_soft_nofp_nosys.cfg -mno-unaligned-access -fshort-enums -fshort-wchar -g -ffunction-sections -fdata-sections -faapcs-bitfield-width -fno-common -Wall -Os`

It comes out that
[image]

voltur01, 2022-01-19:
The file referred to in the error message: https://github.com/kmackay/micro-ecc/blob/master/asm_arm.inc

davemgreen, 2022-01-19:
Hello. Does adding -fomit-frame-pointer help? GCC will have that enabled by default (when not at -O0) where as clang will not, and that will lose you a register.

smithp35, 2022-01-19:
I think the frame pointer is definitely it. Looking at the register trace in clang it r7 is reserved, and this block of inline asm needs every low register. I've checked that `-fomit-frame-pointer` will work, and gcc will not work with `-fno-omit-frame-pointer`

In the source file there is a .syntax divided line. Clang doesn't support that and at least my version gave an error message.

momchil-velikov, 2022-01-19:
Yes, it compiles with `clang -target arm-eabi -march=armv6-m -std=c11 -mno-unaligned-access -fshort-enums -fshort-wchar  -ffunction-sections -fdata-sections -faapcs-bitfield-width -fno-common -Wall -Os -c -fomit-frame-pointer`
`

closed 2022-07-12 by voltur01