ARM-software/LLVM-embedded-toolchain-for-Arm #88: Error when linking with some assembly files

zxtxin, 2022-01-17:
I compiled this two assembly file
[image]
[image]
with the following parameters
'-std=c11 --config armv6m_soft_nofp_nosys.cfg -mno-unaligned-access -fshort-enums -fshort-wchar -g -ffunction-sections -fdata-sections -faapcs-bitfield-width -fno-common -Wall -Os'

But the linker report errors
[image]
with the following parameters
--config armv6m_soft_nofp_nosys.cfg -T /home/wsl/ls_sdk/soc/arm_cm/le501x/compiler/gnu/ble.ld -Wl,--gc-sections

smithp35, 2022-01-19:
This looks like it should be simple to fix. LLD doesn't implement the R_ARM_THM_JUMP8 relocation. I'm a little surprised that the assembler generated the relocation as arm_cm_delay_asm is in the same section. If you need a work-around in the mean-time then I suggest a local symbol at the same offset as arm_cm_delay_asm as that will cause the assembler to resolve the relocation.

Hoping to get some time to send a patch to upstream lld to implement the missing relocation soon.

zxtxin, 2022-01-21:
> This looks like it should be simple to fix. LLD doesn't implement the R_ARM_THM_JUMP8 relocation. I'm a little surprised that the assembler generated the relocation as arm_cm_delay_asm is in the [...]

I

smithp35, 2022-01-21:
Looks like this has already been fixed upstream in https://reviews.llvm.org/D113509 but this isn't in LLD 13.0.0. The 14.0.0 release will include this relocation.

voltur01, 2022-01-21:
Great, thank you for the update!

This means that building the toolchain manually with HEAD (https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/docs/building-from-source.md) should have the fix.

zxtxin, 2022-05-06:
I've tested release 14.0.
It still does not work. But the error log becomes
[image]

smithp35, 2022-05-06:
The first couple of errors are symptomatic of missing the `SHF_ALLOC` flag in a section containing code. A section without `SHF_ALLOC` is not part of the memory that will be seen at run-time, they are used for metadata such as .debug. GNU ld is not as fussy as LLD about this.

The fix for this is to make sure the "ax" flags are included when making a custom ELF section in assembler. For example,
```
 .section mysection, "ax", %progbits
```
Link to GCC .section docs https://sourceware.org/binutils/docs/as/Section.html

The last error message is likely to come from something like:
```
        .text
        .global foo 
        beq foo // conditional branch outside section
```
On most AArch32 Arm CPUs the branch will be a wide branch, but on Cortex-M0 the only relocation is the very short range narrow conditional branch (with the R_ARM_THM_JUMP8) relocation

If foo is too far away from the conditional branch then the linker will give an error message. It looks like the destination is too far in your case. Given the short range of the conditional branch it is not usually practical to use it outside of a section. I suggest replacing it with something like:
```
       .text
        .global foo
        bne  skip // If not equal to then skip
        b foo  // unconditional branch to foo with long range
skip:
        ...
```

zxtxin, 2022-05-06:
> The first couple of errors are symptomatic of missing the `SHF_ALLOC` flag in a section containing code. A section without `SHF_ALLOC` is not part of the memory that will be seen at run-time, they [...]

Thank you.
When I add "ax" flags, all error logs disappear.
Is there any document about the difference between lld and GNU ld?