ARM-software/LLVM-embedded-toolchain-for-Arm #89: Add libraries or new passes to the installation

mintancy, 2022-02-10:
I'm not sure if I understand it correctly.

I want to write a new pass that works on embedded devices, which requires the `opt` tool to load the pass.
For testing, I can simply run `make` directly inside the LLVM repo and use tools inside the `build` folder.
And this is actually what LLVM itself supports.

For other reasons, I need to cross-compile the toolchain and make it work on the Windows system.
Then after building this toolchain, the `install` directory does not contain the tools or libraries that LLVM provides.
I figured out how to add tools to the install/bin folder, such as `opt` and `llvm-link`, by adding new components (`dist_comps`) inside the `make.py` file. 

But I don't know how to add libraries to load passes successfully.
Could you please give me some suggestions? Forgive me if I miss any important instructions. 

I very much appreciate your work on such a wonderful job.

Thank you so much.

smithp35, 2022-02-11:
Hello,

If I've understood correctly:
- You've built your own pass and can run it with the `opt` tool.
- You've altered the scripts so that developer utilities like `opt` are included in the installation

> But I don't know how to add libraries to load passes successfully.
I think that this could mean: 
- how do I alter the build scripts to copy the library that my new pass is in so that it can be loaded with opt?

I don't know how to answer that unfortunately. Our scripts are essentially wrappers around LLVM's build system. I'd expect that you'd need to modify the cmake commands to copy what you need. If you are distributiing a plugin then presumably you are building the pass as a shared library?

I'd like to step back a bit and ask what you're trying to do as there may be alternatives.
- `opt` is a developer tool and it isn't user friendly. I'd recommend that you try and load the pass from clang. This would mean that you don't need to distribute `opt` . I'm not entirely sure how to do this myself with the new pass manager. I believe that there are certain extension points that you can hook into, for example https://stackoverflow.com/questions/54447985/how-to-automatically-register-and-load-modern-pass-in-clang
- I would personally try and avoid creating a plugin if you can and just integrate the pass statically into clang for a number or reasons. The first is that Windows plugin loading is fragile due to the maximum exported symbol limit of a DLL, essentially the LLVM API is often too large to fit into a DLL. The second is that the LLVM API is not stable, so plugins are often tightly coupled to the clang version that they are built against. There is a lot of documentation about building the pass as a static library and inserting it into a pass pipeline. For example https://github.com/banach-space/llvm-tutor https://github.com/alexjung/Writing-an-LLVM-Pass-using-the-new-PassManager . 

> Could you please give me some suggestions? Forgive me if I miss any important instructions.
Sadly this is not well documented especially since the new pass manager has been introduced, it is very easy to miss information. I've put a few links above.

If I've understood you correctly you may be better served by going to the LLVM community directly, work out what to do with a vanilla installation of LLVM and then work out how to change the python scripts to make those changes here.

LLVM has a discourse https://discourse.llvm.org/ and a discord https://discord.gg/xS7Z362 both of which have sections to ask questions like this.

Apologies I can't be more help at the moment.

mintancy, 2022-02-11:
Thank you so much for your reply!
You are right, I should go back to check the LLVM document.
There are a lot of things I need to learn.

closed 2022-02-11 by mintancy