ARM-software/LLVM-embedded-toolchain-for-Arm #107: Does ARM plan to support armv7-a and armv7-r?

xiaoxiang781216, 2022-04-15:


voltur01, 2022-04-19:
Hi Xiang, yes, we plan to add A-profile and R-profile support to the toolchain, probably, after https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues/110 is addressed so that it is easier to manage the amount of required library variants.

xiaoxiang781216, 2022-04-19:
Thanks, when will it be supported?

voltur01, 2022-04-19:
We are reviewing our roadmap right now, so unfortunately I do not have dates yet, one thing is sure that this is part of long term planning, i.e. this will not happen in coming weeks or even months.

voltur01, 2023-07-26:
Hi, we may have some basic Armv7-A and Armv7-R configurations provided in time for the LLVM 17 based release, so in October, it is on the top of the backlog, however no promises yet.

voltur01, 2025-01-06:
The set of library variants was significantly extended, so we will have basic A, R, and M-profile coverage in the upcoming LLVM 20 based release.

closed 2025-01-06 by voltur01