ARM-software/LLVM-embedded-toolchain-for-Arm #111: Plan to provide support Single-precision(-mfpu=fpv5-sp-d16) on Cortex-M33

anchao, 2022-04-19:
Hello ARM,

I've been integrating the Clang compiler recently, seems Clang Tool chain only provide the double-precision(-mfpu=fpv5-d16) library in the configuration of armv8-m.main(Cortex-M33), Does ARM plan to provide support for multi libraries with different configurations?

```
$ clang  --config armv8m.main_hard_fp_nosys --print-file-name libclang_rt.builtins-armv8m.main.a
lib/clang-runtimes/armv8m.main_hard_fp/lib/libclang_rt.builtins-armv8m.main.a
```

```
00000000 <__divdc3>:
   0: e92d 43f0   stmdb sp!, {r4, r5, r6, r7, r8, r9, lr}
...
  16: eeb0 0bc3   vabs.f64  d0, d3
  1a: eeb0 1bc4   vabs.f64  d1, d4
  1e: eeb4 1b40   vcmp.f64  d1, d0
...
```

voltur01, 2022-04-26:
Hi,

We do not expect to add many new configurations before we have multilib support (https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues/110) as a better way to manage them.

Meanwhile, it should be possible to add your own configurations by extending `config.py`, see https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/cc15bcde0b7e295fe8e531c6fc57782b6accb609/scripts/config.py#L221

anchao, 2022-05-05:
> Hi,
>
> We do not expect to add many new configurations before we have multilib support (#110) as a better way to manage them.
>
> Meanwhile, it should be possible to add your own configurations by [...]

Thanks for reply, seems multi-library support is already in your plans, let me close this PR

closed 2022-05-05 by anchao