ARM-software/LLVM-embedded-toolchain-for-Arm #112: about command line length limit in windows

versaloon, 2022-07-27:
Command Line:
G:\project\vsf\example\template\project\cmake\aic8800\build>E:\Software\armllvm\bin\clang.exe -mcpu=cortex-m4 -mthumb --config armv6m_soft_nofp_rdimon -Ofast -std=gnu11 -Wall -fno-builtin -fno-strict-aliasing -fdata-sections -fms-extensions -ffunction-sections -fms-extensions -fdiagnostics-color=always -Xlinker -TG:/project/vsf/example/template/project/cmake/aic8800/linker.ld -Xlinker -Map=output.map -mabi=aapcs -march=armv7e-m @CMakeFiles\vsf_template.rsp -o vsf_template.elf -Wl,--out-implib,libvsf_template.dll.a -Wl,--major-image-version,0,--minor-image-version,0
clang: error: unable to execute command: Couldn't execute program 'E:/Software/armllvm/bin/ld.lld.exe': 文件名或扩展名太长。  (0xCE)
clang: error: ld.lld command failed with exit code 1 (use -v to see invocation)

It seems that the problem is because of the command line length limit in win10, is it possible to fix this?
There are 700 .c files in the project, so it will make the link command too long in windows.

voltur01, 2022-07-27:
Hello,

Thank you for trying the toolchain and reporting the issue!

The recommended way to overcome the command line length limitation is to use [Response files](https://llvm.org/docs/CommandLine.html#response-files).

versaloon, 2022-07-28:
> Hello,
> 
> Thank you for trying the toolchain and reporting the issue!
> 
> The recommended way to overcome the command line length limitation is to use [Response files](https://llvm.org/docs/CommandLine.html#response-files).

Thanks

closed 2022-07-28 by versaloon