ARM-software/LLVM-embedded-toolchain-for-Arm #129: Provide a flavor with exception support

KKoovalsky, 2022-09-04:
Hello!

First, I am really excited that this project is being developed, since other developers might not have to cross-compile the compiler runtime, libc, libc++, ... on their own. Moreover, having an alternative to the ARM GNU Toolchain based on LLVM is so cool!

However, I cross-compile firmware with the exceptions enabled, but the main page lists the exception handling feature as not supported. The ARM GNU Toolchain provided c++ libraries in _nano_ and "normal" flavor. Linking to the latter flavor, exceptions were handled correctly.

Could this project provide something similar? The toolchain could bundle libc++ ABI library in _nano_ and _exceptions_ flavors.

voltur01, 2022-09-05:
Hi Kacper,

Thank you for your interest in the toolchain!

Yes, we plan to enable exceptions support as well, however we want to sort https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues/110 out first so that it is easier to manage the additional library variants. There is no date we can promise, but it is coming soon.

voltur01, 2024-03-22:
Library variants with exceptions support provided since https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/398

closed 2024-03-22 by voltur01