ARM-software/LLVM-embedded-toolchain-for-Arm #157: Cannot compile simple c++ program

erectcrested, 2022-11-22:
Hi.
I am trying to compile simple c++ program with semihosting without success.  My cmdline is:
```
clang++ -o hello hello.cpp --config aarch64_semihost.cfg -T picolibcpp.ld -fuse-ld=lld
```

The program is:
```
#include <iostream>
using namespace std;

int main(int argc, char *argv[])
{
    cout << "Hello world!" << endl;
    return 0;
}                                                                                                                                                                                                                                                                                                                                                                                                               
``` 

Since few commits ago it localization was not enabled so iostream was not usable at all. After localization was enabled I got some errors regarding wide characters support in iostream. 

After changing `-DLIBCXX_ENABLE_WIDE_CHARACTERS=OFF` to `-DLIBCXX_ENABLE_WIDE_CHARACTERS=ON` in CMakeLists.txt I am not getting the following errors:

> clang++ -o hello hello.cpp --config aarch64_semihost.cfg -T picolibcpp.ld -fuse-ld=lld
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 948 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 1756 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 1764 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 1824 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 1956 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 2124 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 2304 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 2472 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 2648 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 2816 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 2912 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 4048 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 4144 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 4236 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 4332 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 4424 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 4520 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 4612 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 4808 bytes
> ld.lld: error: section '.text' will not fit in region 'flash': overflowed by 5032 bytes
> ld.lld: error: too many errors emitted, stopping now (use --error-limit=0 to see all errors)
> clang-16: error: ld.lld command failed with exit code 1 (use -v to see invocation)

mplatings, 2022-11-22:
> After localization was enabled I got some errors regarding wide characters support in iostream

It's possible your llvm-project checkout isn't up to date.

`LIBCXX_ENABLE_WIDE_CHARACTERS=ON` is not supported. However the errors you pasted don't look related to that. You might to change your linker script. See [Using picolibc.ld](https://github.com/picolibc/picolibc/blob/main/doc/linking.md#using-picolibcld).

See [this test](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/tests/smoketests/cpp-basic-semihosting/hello.cpp) for an example of how to build a Hello World.

erectcrested, 2022-11-22:
Is there a reason why `LIBCXX_ENABLE_WIDE_CHARACTERS=ON` don't work?
I have juts now used the link you sent me and I saw you use microbit.ld script which is a wrapper around picolibc.ld script, just that it adds all the nesseccary sections location and sizes for: _flash and _ram. 
So by using `-T microbit.ld` which I copied from [here](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/samples/ldscripts/microbit.ld) instead of `-T picolibcpp.ld`, the compilation finished successfully. 

My toolchain is compiled with `LIBCXX_ENABLE_WIDE_CHARACTERS=ON`. Would it cause any problems?

closed 2022-11-22 by mplatings