ARM-software/LLVM-embedded-toolchain-for-Arm #168: Build failure on macOS: FAILED: lib/libunwind.a

JetForMe, 2022-12-30:
It's not clear to me exactly what's gone wrong here. It fails on `ninja llvm-toolchain`:

```
-- Build files have been written to: /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/runtimes/armv6m_soft_nofp/src/runtimes_armv6m_soft_nofp-build
[3042/3236] Performing build step for 'runtimes_aarch64'
[298/921] Building CXX object libunwind/src/CMakeFiles/unwind_static_objects.dir/libunwind.cpp.o
In file included from /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/_deps/llvmproject-src/libunwind/src/libunwind.cpp:30:
/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/_deps/llvmproject-src/libunwind/src/AddressSpace.hpp:486:58: warning: unused parameter 'targetAddr' [-Wunused-parameter]
inline bool LocalAddressSpace::findUnwindSections(pint_t targetAddr,
                                                         ^
1 warning generated.
[312/921] Linking C static library lib/libunwind.a
FAILED: lib/libunwind.a 
: && "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool" -static -no_warning_for_no_symbols -o lib/libunwind.a  libunwind/src/CMakeFiles/unwind_static_objects.dir/libunwind.cpp.o libunwind/src/CMakeFiles/unwind_static_objects.dir/Unwind-EHABI.cpp.o libunwind/src/CMakeFiles/unwind_static_objects.dir/Unwind-seh.cpp.o libunwind/src/CMakeFiles/unwind_static_objects.dir/Unwind_AppleExtras.cpp.o libunwind/src/CMakeFiles/unwind_static_objects.dir/UnwindLevel1.c.o libunwind/src/CMakeFiles/unwind_static_objects.dir/UnwindLevel1-gcc-ext.c.o libunwind/src/CMakeFiles/unwind_static_objects.dir/Unwind-sjlj.c.o libunwind/src/CMakeFiles/unwind_static_objects.dir/UnwindRegistersRestore.S.o libunwind/src/CMakeFiles/unwind_static_objects.dir/UnwindRegistersSave.S.o && :
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/libunwind.cpp.o is not an object file (not allowed in a library)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/Unwind-EHABI.cpp.o is not an object file (not allowed in a library)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/Unwind-seh.cpp.o is not an object file (not allowed in a library)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/Unwind_AppleExtras.cpp.o is not an object file (not allowed in a library)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/UnwindLevel1.c.o is not an object file (not allowed in a library)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/UnwindLevel1-gcc-ext.c.o is not an object file (not allowed in a library)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/Unwind-sjlj.c.o is not an object file (not allowed in a library)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/UnwindRegistersRestore.S.o is not an object file (not allowed in a library)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: libunwind/src/CMakeFiles/unwind_static_objects.dir/UnwindRegistersSave.S.o is not an object file (not allowed in a library)
[323/921] Copying CXX header __format/formatter_integer.h
ninja: build stopped: subcommand failed.
FAILED: runtimes/aarch64/src/runtimes_aarch64-stamp/runtimes_aarch64-build /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/runtimes/aarch64/src/runtimes_aarch64-stamp/runtimes_aarch64-build 
cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/runtimes/aarch64/src/runtimes_aarch64-build && /opt/homebrew/Cellar/cmake/3.25.1/bin/cmake --build .
ninja: build stopped: subcommand failed.
```

macOS: 13.1
Xcode: 14.1
```
$ clang --version
Homebrew clang version 15.0.6
Target: arm64-apple-darwin22.2.0
Thread model: posix
InstalledDir: /opt/homebrew/opt/llvm/bin
```

I also get a number of CMake warnings like

```
CMake Warning:
  Manually-specified variables were not used by the project:

    LIBCXX_ENABLE_DEBUG_MODE_SUPPORT
    LLVM_CCACHE_BUILD
```

mplatings, 2022-12-30:
Yeah, runtimes don't seem to build on Mac. I think a fix is needed in llvm-project but you can have a look at #156 and see if that works for you.

JetForMe, 2022-12-30:
My `llvm-HEAD.patch` seems quite different from the one that PR changes. Mine contains only this:

```
diff --git a/.gitignore b/.gitignore
index 20c4f52cd378..eefdedbcb4f4 100644
--- a/.gitignore
+++ b/.gitignore
@@ -1,2 +1,2 @@
-#==============================================================================#
+#==============================================================================#
 # This file specifies intentionally untracked files that git should ignore.
```

mplatings, 2022-12-30:
Yep, since that PR was created llvm-HEAD.patch was changed to do nothing.

JetForMe, 2022-12-31:
Okay, I don't know nearly enough about PRs and patch files to know how to modify my own build. I found two `runtimes/CMakeLists.txt` files in the stuff checked out so far:

```
_deps/llvmproject-src/runtimes/CMakeLists.txt
_deps/llvmproject-src/llvm/runtimes/CMakeLists.txt
```

Only one has `Runtimes C CXX ASM` in it (`_deps/llvmproject-src/runtimes/CMakeLists.txt`), so I added this as line 12 in the file:

```
set(APPLE False)
```

And that seemed to allow the build to complete.

JetForMe, 2022-12-31:
Not sure if you'd prefer a new issue, or to keep going here. Now some of the tests fail:

```
24285 17:22: build/ (main) $ ninja check-llvm-toolchain
...
[107/115] Performing install step for 'runtimes_armv8.1m.main_hard_nofp_mve'
-- Install configuration: "MinSizeRel"
[108/115] Performing install step for 'runtimes_armv8.1m.main_soft_nofp_nomve'
-- Install configuration: "MinSizeRel"
[112/115] cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/samples/src/baremetal-semihosting && make clean && make run BIN_PATH=/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin && make clea
FAILED: CMakeFiles/check-llvm-toolchain-samples-baremetal-semihosting /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/CMakeFiles/check-llvm-toolchain-samples-baremetal-semihosting 
cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/samples/src/baremetal-semihosting && make clean && make run BIN_PATH=/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin && make clean
rm -f *.elf *.hex
/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin/clang --config armv6m_soft_nofp_semihost.cfg -g -T ../../ldscripts/microbit.ld -o hello.elf hello.c
/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin/llvm-objcopy -O ihex hello.elf hello.hex
qemu-system-arm -M microbit -semihosting -nographic -device loader,file=hello.hex
make: qemu-system-arm: No such file or directory
make: *** [run] Error 1
[113/115] cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/tests/smoketests/basic-semihosting && ma...ean && make run BIN_PATH=/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin && make clean
FAILED: CMakeFiles/check-llvm-toolchain-smoketests-basic-semihosting /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/CMakeFiles/check-llvm-toolchain-smoketests-basic-semihosting 
cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/tests/smoketests/basic-semihosting && make clean && make run BIN_PATH=/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin && make clean
rm -f hello.elf hello.hex
make: qemu-system-arm: No such file or directory
make: *** [run] Error 1
[114/115] cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/samples/src/cpp-baremetal-semihosting &&...ean && make run BIN_PATH=/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin && make clean
FAILED: CMakeFiles/check-llvm-toolchain-samples-cpp-baremetal-semihosting /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/CMakeFiles/check-llvm-toolchain-samples-cpp-baremetal-semihosting 
cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/samples/src/cpp-baremetal-semihosting && make clean && make run BIN_PATH=/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin && make clean
rm -f *.elf *.hex
/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin/clang++ --config armv6m_soft_nofp_semihost.cfg -g -T ../../ldscripts/microbit.ld -o hello.elf hello.cpp
/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin/llvm-objcopy -O ihex hello.elf hello.hex
qemu-system-arm -M microbit -semihosting -nographic -device loader,file=hello.hex
make: qemu-system-arm: No such file or directory
make: *** [run] Error 1
[115/115] cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/tests/smoketests/cpp-basic-semihosting &...ean && make run BIN_PATH=/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin && make clean
FAILED: CMakeFiles/check-llvm-toolchain-smoketests-cpp-basic-semihosting /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/CMakeFiles/check-llvm-toolchain-smoketests-cpp-basic-semihosting 
cd /Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/tests/smoketests/cpp-basic-semihosting && make clean && make run BIN_PATH=/Users/rmann/Projects/Personal/EmbeddedSwift/LLVM-embedded-toolchain-for-Arm/build/llvm/bin && make clean
rm -f hello.elf hello.hex
make: qemu-system-arm: No such file or directory
make: *** [run] Error 1
ninja: build stopped: subcommand failed.
```

mplatings, 2023-01-03:
That failure is likely due to you not having QEMU installed. Probably everything is fine.

stuij, 2023-02-24:
For more visibility, I made a pull request a few months back with the same fix that made things work for @JetForMe: https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/156

I since updated it to work with more recent changes. Haven't tested it recently on HEAD.

stuij, 2023-02-24:
Just updated https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/156 to work with HEAD again.

stuij, 2023-02-24:
ah right, my bad. reading back the comments properly I see that @michaelplatingsarm referenced my pull request and that's why the solutions were the same :) Anyways, for who reads this, hope the update helps.

mplatings, 2023-02-28:
Fixed by #188

closed 2023-02-28 by mplatings