ARM-software/LLVM-embedded-toolchain-for-Arm #179: Release tar extracts all files into current folder instead of a new one

DavidSpickett, 2023-02-07:
I went to https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases/tag/release-15.0.2 and downloaded https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases/download/release-15.0.2/LLVMEmbeddedToolchainForArm-15.0.2-Linux-AArch64.tar.gz.

Extracted it with:
```
tar -xvf <file>
```
And ended up with a bunch of new folders and files in the current dir, as opposed to one folder `LLVMEmbeddedToolchainForArm-15.0.2-Linux-AArch64/...`. The former is unexpected and a pain to clean up.
```
$ ls
CHANGELOG.md  LLVMEmbeddedToolchainForArm-15.0.2-Linux-AArch64.tar.gz  THIRD-PARTY-LICENSES.txt  bin   include  samples  third-party-licenses
LICENSE.txt   README.md                                                VERSION.txt               docs  lib      share
```

I could probably find a tar flag that prevents this but it's not a problem with other toolchain releases e.g. llvm's. Can we adjust the packaging to avoid this?

https://github.com/llvm/llvm-project/releases/download/llvmorg-14.0.6/clang+llvm-14.0.6-aarch64-linux-gnu.tar.xz if you need to look at one that does this correctly. Though I guess it's a matter of compressing a folder, rather than the contents of the folder.

DavidSpickett, 2023-02-07:
For what it's worth this release's archives behave as expected https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases/tag/release-14.0.0.

voltur01, 2023-02-07:
Hi David,

Thank you for highlighting this! I added corresponding issue to the team backlog.

mplatings, 2023-02-07:
This is fixed on the main branch, and will be fixed in the 16.0.0 release

closed 2023-02-07 by mplatings