ARM-software/LLVM-embedded-toolchain-for-Arm #273: Is it possible to generate ram assess codes by static base register(r9) instead of PC relative for position independency?

versaloon, 2023-07-16:
I'm implementing a XIP elfloader for CortexM series, the .text/.ro will be in flash and .bss/.data will be in sram.

There are 2 options for position independency:
1. embedded position independency by -fropi -frwpi
But I found some .bss/.data access code will use PC-relative form, which will cause hardfault. Because .text is in flash and .bss/.data is in ram, so the offset calculated on virtual address is not fit.
3. GOT-based position independency by -fPIC
But it seems -mno-pic-data-is-text-relative is not supported by armv7m.

versaloon, 2023-07-16:
It seems a busybox optimization not suitable for CortexM4.

/* We use a trick to have more optimized code (fewer pointer reloads
 * and reduced binary size by a few kilobytes) like:
 *  ash.c:   extern struct globals *const ash_ptr_to_globals;
 *  ash_ptr_hack.c: struct globals *ash_ptr_to_globals;
 * This way, compiler in ash.c knows the pointer can not change.
 *
 * However, this may break on weird arches or toolchains. In this case,
 * set "-DBB_GLOBAL_CONST=''" in CONFIG_EXTRA_CFLAGS to disable
 * this optimization.
 */
#ifndef BB_GLOBAL_CONST
# define BB_GLOBAL_CONST const
#endif

For CortexM4, if const is used, the variable will be assessed with pc relative code.

closed 2023-07-16 by versaloon

smithp35, 2023-07-17:
> embedded position independency by -fropi -frwpi
But I found some .bss/.data access code will use PC-relative form, which will cause hardfault. Because .text is in flash and .bss/.data is in ram, so the offset calculated on virtual address is not fit.

With -fropi and -frwpi only read-only data should be accessed PC-relative. The intent is that the .rodata is kept in the same "segment" as the code. For example:
```
-----------
RO-CODE
RO-DATA
-----------

r9
-----------
RW-DATA
BSS
-----------
```
It is a bug if RW/.bss data is being accessed pc-relative with -fropi/-frwpi. However it is expected that read-only data is accessed this way.

If I interpret
```
#ifndef BB_GLOBAL_CONST
# define BB_GLOBAL_CONST const
#endif
```
Then this would make whatever BB_GLOBAL_CONST read-only data and would mean it was accessed pc-relative.

You are correct that clang doesn't support -mno-pic-data-is-text-relative which is a pity as while -fropi and -frwpi work well for the programs they support, they do have limitations.