ARM-software/LLVM-embedded-toolchain-for-Arm #277: Support for softfp?

adityashah1212, 2023-07-24:
Seems like the packaged versions of standard library are either `hard` or `soft`, there doesn't seem to be any packaged libraries for `softfp` variants. Is this deliberate or was simply missed out?

mplatings, 2023-07-24:
It's deliberate. Why do you want use `softfp`?

adityashah1212, 2023-07-24:
Some vendors provide libraries that use `softfp` abi. [Silicon Labs Gecko SDK](https://github.com/SiliconLabs/gecko_sdk) is a good example of that (though they don't currently support LLVM toolchain). I imagine, since they are different ABI's, we won't be able to link those libraries with `hard` float standard libraries.

smithp35, 2023-07-25:
Our thoughts were that softfp was unlikely to be used by many people, with the remaining use case someone that could guarantee that they had a hard floating point unit, but needed to link against some legacy binary only soft, or softfp libraries that they could not rebuild. In an open source context we thought that the use case was small.

adityashah1212, 2023-07-26:
Well, in fully open source context I agree that use case is small, but there are a lot of use cases where users will rely on vendor provided "non-rebuildable" libraries while still using `LLVM` or `gcc` compilers. IMO, in such scenarios, whether the project itself is open-source or proprietary is hardly relevant.

mplatings, 2023-07-27:
There's always going to be the problem that the finite set of libraries provided by the toolchain won't match a particular set of options. I think the toolchain should provide enough libraries that there will be a variant that is _compatible_, but not necessarily _optimal_. In the case of `softfp`, the `soft` library variants provide that compatibility.

However clearly performance is extremely important. To address this, I think that instead of trying to package more and more library variants, the toolchain should make it really easy to build your own system libraries.

Currently the way to build your own libraries is to check out the LLVM Embedded Toolchain for Arm sources, patch [the library variant list in CMakeLists.txt](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/CMakeLists.txt#L913) and build the entire toolchain yourself. Clearly there's some way to go between the current state and "really easy".

One idea I have of "really easy" is, in your project's CMakeLists.txt, add something like:
`add_subdirectory("${LLVM_EMBEDDED_TOOLCHAIN_FOR_ARM}/library-source" SYSTEM)`
Then, no matter how obscure your compilation options might be, the system libraries will automatically match exactly. That's the theory anyway.

voltur01, 2023-08-31:
A note added to the docs on how to add own library variants https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/299

voltur01, 2023-09-11:
`softfp` library variant added by https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/302

closed 2023-09-11 by voltur01