ARM-software/LLVM-embedded-toolchain-for-Arm #281: Cannot select: intrinsic %llvm.arm.mve.vcvta

finchky, 2023-07-27:
Using v16.0.0

```
#0 0x000055c5ce8e51da llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x22e51da)
 #1 0x000055c5ce8e2f64 llvm::sys::CleanupOnSignal(unsigned long) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x22e2f64)
 #2 0x000055c5ce844dd3 llvm::CrashRecoveryContext::HandleExit(int) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2244dd3)
 #3 0x000055c5ce8da06e llvm::sys::Process::Exit(int, bool) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x22da06e)
 #4 0x000055c5cd41dcb6 (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0xe1dcb6)
 #5 0x000055c5ce84c54f llvm::report_fatal_error(llvm::Twine const&, bool) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x224c54f)
 #6 0x000055c5cf8681a4 llvm::SelectionDAGISel::CannotYetSelect(llvm::SDNode*) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x32681a4)
 #7 0x000055c5cf86c012 llvm::SelectionDAGISel::SelectCodeCommon(llvm::SDNode*, unsigned char const*, unsigned int) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x326c012)
 #8 0x000055c5cd7cd107 (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x11cd107)
 #9 0x000055c5cf8667bc llvm::SelectionDAGISel::DoInstructionSelection() (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x32667bc)
#10 0x000055c5cf86fce6 llvm::SelectionDAGISel::CodeGenAndEmitDAG() (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x326fce6)
#11 0x000055c5cf8730ca llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x32730ca)
#12 0x000055c5cf874bea (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x3274bea)
#13 0x000055c5cd7d6124 (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x11d6124)
#14 0x000055c5cdd0e432 (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x170e432)
#15 0x000055c5ce21bf29 llvm::FPPassManager::runOnFunction(llvm::Function&) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x1c1bf29)
#16 0x000055c5ce21c1c1 llvm::FPPassManager::runOnModule(llvm::Module&) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x1c1c1c1)
#17 0x000055c5ce21cc57 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x1c1cc57)
#18 0x000055c5cec67981 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::HeaderSearchOptions const&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, std::unique_ptr<llvm::raw_pwrite_stream, std::default_delete<llvm::raw_pwrite_stream>>) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2667981)
#19 0x000055c5cf9a66dd (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x33a66dd)
#20 0x000055c5d06bba09 clang::ParseAST(clang::Sema&, bool, bool) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x40bba09)
#21 0x000055c5cf9a71ed clang::CodeGenAction::ExecuteAction() (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x33a71ed)
#22 0x000055c5cf35f281 clang::FrontendAction::Execute() (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2d5f281)
#23 0x000055c5cf2f3f9a clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2cf3f9a)
#24 0x000055c5cf432ab3 clang::ExecuteCompilerInvocation(clang::CompilerInstance*) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2e32ab3)
#25 0x000055c5cd41f6b4 cc1_main(llvm::ArrayRef<char const*>, char const*, void*) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0xe1f6b4)
#26 0x000055c5cd41a5bf (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0xe1a5bf)
#27 0x000055c5cf1729a5 (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2b729a5)
#28 0x000055c5ce844cd5 llvm::CrashRecoveryContext::RunSafely(llvm::function_ref<void ()>) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2244cd5)
#29 0x000055c5cf175178 clang::driver::CC1Command::Execute(llvm::ArrayRef<std::optional<llvm::StringRef>>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>*, bool*) const (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2b75178)
#30 0x000055c5cf1421ab clang::driver::Compilation::ExecuteCommand(clang::driver::Command const&, clang::driver::Command const*&, bool) const (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2b421ab)
#31 0x000055c5cf142cf3 clang::driver::Compilation::ExecuteJobs(clang::driver::JobList const&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&, bool) const (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2b42cf3)
#32 0x000055c5cf149bdf clang::driver::Driver::ExecuteCompilation(clang::driver::Compilation&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0x2b49bdf)
#33 0x000055c5cd41d491 clang_main(int, char**) (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0xe1d491)
#34 0x00007fc439c49083 __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x24083)
#35 0x000055c5cd41578a _start (/opt/LLVMEmbeddedToolchainForArm-16.0.0-Linux-x86_64/bin/clang-16+0xe1578a)
clang-16: error: clang frontend command failed with exit code 70 (use -v to see invocation)
```

[SupportFunctions-ae2af8.zip](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/files/12188098/SupportFunctions-ae2af8.zip)

mplatings, 2023-07-27:
That sounds like it will be an issue in clang.

You could open a bug in https://github.com/llvm/llvm-project/issues but the chances of it being looked at will be much improved if you reduce the reproducer to the bare minimum to reproduce the crash e.g. 50 lines instead of 50,000 lines.

closed 2023-07-27 by mplatings

davemgreen, 2023-07-28:
To attempt to save you a trip, I think the -mfpu option is causing problems. Using -mfpu=fpv5-sp-d16 is essentially disabling mve in the backend, so it can't select any of the intrinsics and scalarized the vector instructions. There is a disconnect between the frontend (clang) and the backend (llvm) about what is enabled, where it should probably be producing better error messages.

Using -mcpu=cortex-m85 on it's own (without a -mfpu) should be enough to pick a good default fpu for the cpu. Otherwise it likely needs to use -mfpu=fp-armv8-fullfp16-d16 instead.

finchky, 2023-07-28:
> To attempt to save you a trip, I think the -mfpu option is causing problems. Using -mfpu=fpv5-sp-d16 is essentially disabling mve in the backend, so it can't select any of the intrinsics and [...]

Omitting `-mfpu` altogether gave me some other errors. But specifying `-mfpu=fp-armv8-fullfp16-sp-d16` seemed to have worked. Thanks!

Is this potentially an issue with the multilib mapping? I read that is still in experimental support with 16.0.0

voltur01, 2023-07-31:
@finchky thanks for the confirmation!

If you could provide a bit more details around "Omitting -mfpu altogether gave me some other errors", it would make it easier for us to decide if there is anything to fix here.

finchky, 2023-07-31:
@voltur01 sure!

My original compile command was 
```
clang -std=c99 -x c -mfloat-abi=hard -mfpu=fpv5-sp-d16 -O2 -w -g3 -gdwarf-4 -mcpu=cortex-m85 --target=arm-none-eabi -mthumb -I ra/arm/CMSIS_5/CMSIS/Core/Include -I ra/arm/CMSIS_5/CMSIS/DSP/Include -I ra/arm/CMSIS_5/CMSIS/DSP/Examples/ARM/arm_fir_example -I ra/arm/CMSIS_5/CMSIS/DSP/PrivateInclude -c  -o SupportFunctions.o" SupportFunctions.c
```

Following Dave's suggestion, I then tried the above command without the `-mpfu=fpv5-sp-d16` argument. This gave me errors that `setjmp.h` could not be found from one of our internal source files. It also gave that `string.h` could not be found from one of the DSP source files:
[image]

However, when I replaced `-mpfu=fpv5-sp-d16` from the above command with `-mfpu=fp-armv8-fullfp16-sp-d16`, per Dave's suggestion, I was able to compile successfully.

mplatings, 2023-07-31:
Yes, the `'string.h' file not found` is related to multilib. No multilib matched those settings.

This will be improved in 2 ways in LLVM Embedded Toolchain for Arm 17:
1. FPU matching is greatly improved, so a suitable multilib would be found for your flags without `-mfpu=fp-armv8-fullfp16-sp-d16`.
2. If no multilib matches a given set of flags, clang will emit a warning in such cases: `warning: no multilib found matching flags: ...`. (However the flags you specified included `-w` so the warning would be suppressed in that case.)