ARM-software/LLVM-embedded-toolchain-for-Arm #295: Is armv7em + FPv5-SP-D16-M supported?

andreiion, 2023-08-11:
Hei everyone,

I am trying to use the toolchain to build a specific configuration for FPU: `-mfpu=fpv5-sp-d16`. More exactly, I am trying to have single-precision.

According to documentation [here][1], both single and double precision are included in `v7E-M + FPv5-DP-D16-M`.
Is this correct?

In the repo, I noticed that there is only one `.cfg` file for `fpv5`, omitting `DP` or `SP` (`armv7em_hard_fpv5_d16.cfg`).
Moreover, the runtime libs only support this version `./lib/clang-runtimes/armv7em_hard_fpv5_d16`. 

My questions would be:
Can `armv7em_hard_fpv5_d16` libraries be used for both single and double precision? Can I use, for example, `/lib/clang-runtimes/armv7em_hard_fpv5_d16/lib/libclang_rt.builtins-armv7em.a` from this repo for `-mfpu=fpv5-sp-d16`?

[1]: https://developer.arm.com/documentation/ddi0489/f/floating-point-unit/about-the-fpu?lang=en

mplatings, 2023-08-11:
Hi Andrei,

If `-mfpu=fpv5-sp-d16` is the configuration you want then `armv7em_hard_fpv5_d16` would not be appropriate because it needs double-precision floating point hardware.

According to the documentation you linked to, the Cortex-M7 is available in three configurations, which may or may not include single or double-precision floating point hardware.

For LLVM Embedded Toolchain for Arm 16 and later you don't have to choose a config file. Instead the multilib system will choose a suitable library based on your flags. See https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/tree/llvm-16#experimental-multilib.

For `-mfpu=fpv5-sp-d16` it looks like the multilib system would choose the `armv7em_hard_fpv4_sp_d16` library, which I think is the best choice for you out of the available options.

closed 2023-08-11 by mplatings

andreiion, 2023-08-11:
Hei Michael,

Thank you for a prompt reply!

I will give it a try with fpv4.
According to [this][1] page as well, there doesn't seem to be much difference between `fpv4` and `fpv5` that I should care about :)

[1]: https://developer.arm.com/documentation/ddi0489/f/floating-point-unit/about-the-fpu