ARM-software/LLVM-embedded-toolchain-for-Arm #307: Simple C++ program cannot seem to find stdint.h for Cortex-M85

finchky, 2023-09-12:
I am trying to compile a simple program:

```
#include <stdint.h>

int main(void)
{
    uint8_t count = 0;
    uint32_t result = 0;

    for (count = 0; count < 10; count++)
    {
        result += count;
    }

    return result;
}
```

with command:

```
clang -std=c++11 -x c++ -mfloat-abi=hard -O2 -mcpu=cortex-m85 --target=arm-none-eabi -mthumb -c -o test.o test.cpp
```

If I change the above command to `mcpu=cortex-m33` or `mcpu=cortex-m4` it works fine - no errors. But with the above command, I get the following output:

```
clang version 17.0.0
Target: arm-none-unknown-eabi
Thread model: posix
InstalledDir: /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin
 (in-process)
 "/opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/clang-17" -cc1 -triple thumbv8.1m.main-none-unknown-eabihf -emit-obj -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name test.cpp -mrelocation-model static -mframe-pointer=all -fmath-errno -ffp-contract=on -fno-rounding-math -mconstructor-aliases -nostdsysteminc -target-cpu cortex-m85 -target-feature -crc -target-feature -dotprod -target-feature +dsp -target-feature +mve -target-feature +mve.fp -target-feature +ras -target-feature -fp16fml -target-feature -bf16 -target-feature -sb -target-feature -i8mm -target-feature +lob -target-feature -cdecp0 -target-feature -cdecp1 -target-feature -cdecp2 -target-feature -cdecp3 -target-feature -cdecp4 -target-feature -cdecp5 -target-feature -cdecp6 -target-feature -cdecp7 -target-feature +pacbti -target-feature -hwdiv-arm -target-feature +hwdiv -target-feature +vfp2 -target-feature +vfp2sp -target-feature -vfp3 -target-feature +vfp3d16 -target-feature +vfp3d16sp -target-feature -vfp3sp -target-feature +fp16 -target-feature -vfp4 -target-feature +vfp4d16 -target-feature +vfp4d16sp -target-feature -vfp4sp -target-feature -fp-armv8 -target-feature +fp-armv8d16 -target-feature +fp-armv8d16sp -target-feature -fp-armv8sp -target-feature +fullfp16 -target-feature +fp64 -target-feature -d32 -target-feature -neon -target-feature -crypto -target-feature -sha2 -target-feature -aes -target-feature +strict-align -target-abi aapcs -mfloat-abi hard -Wunaligned-access -debugger-tuning=gdb -v -fcoverage-compilation-dir=/home/coder/workspace/peaks -resource-dir /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/lib/clang/17 -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv8.1m.main_hard_fp/include/c++/v1 -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv8m.main_hard_fp/include/c++/v1 -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv5_d16/include/c++/v1 -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv4_sp_d16/include/c++/v1 -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/lib/clang/17/include -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv8.1m.main_hard_fp/include -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv8m.main_hard_fp/include -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv5_d16/include -internal-isystem /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv4_sp_d16/include -O2 -std=c++11 -fdeprecated-macro -fdebug-compilation-dir=/home/coder/workspace/peaks -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcxx-exceptions -fexceptions -fcolor-diagnostics -vectorize-loops -vectorize-slp -faddrsig -D__GCC_HAVE_DWARF2_CFI_ASM=1 -o test.o -x c++ test.cpp
clang -cc1 version 17.0.0 based upon LLVM 17.0.0-rc1 default target aarch64-linux-gnu
ignoring duplicate directory "/opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/lib/clang/17/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv8.1m.main_hard_fp/include/c++/v1
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv8m.main_hard_fp/include/c++/v1
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv5_d16/include/c++/v1
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv4_sp_d16/include/c++/v1
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/lib/clang/17/include
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv8.1m.main_hard_fp/include
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv8m.main_hard_fp/include
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv5_d16/include
 /opt/LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv4_sp_d16/include
End of search list.
test.cpp:5:5: error: unknown type name 'uint8_t'
    5 |     uint8_t count = 0;
      |     ^
test.cpp:6:5: error: unknown type name 'uint32_t'
    6 |     uint32_t result = 0;
      |     ^
2 errors generated.
```

smithp35, 2023-09-13:
I know what the problem is, but I don't have a good solution for it right now. It may need a rethink of how multilib works unfortunately.

There are multiple libraries that are compatible with the chosen variant. Hence the long list of C++ include directories. Unfortunately the C++ library has a <stdint.h> which uses #include_next expecting this to find the C libraries <stdint.h> however as there are 3 more C++ library includes then it will find <stdint.h> from there and fail to find the original.

Not got a good answer to this one right now, will need to have a think.

voltur01, 2023-09-13:
Thank you for the analysis, Peter!

As a reminder, `--sysroot` can be pointed directly to the library folder to bypass multilibs logic, e.g. to unblock testing before a multilibs solution is available.

PR #312 (2023-09-28): add ExclusiveGroup feature to multilib.yaml
Fixes #307
This allows a multilib configuration to specify explicitly that a subset of its
libraries are alternatives to each other, and at most one of that subset should
be selected. To be consistent with existing semantics, the _last_ choice wins,
if more than one of the options matches the given flags.
closed 2023-09-28 by amilendra