ARM-software/LLVM-embedded-toolchain-for-Arm #331: Distribution via vcpkg requires zip/tarball archives

JonatanAntoni, 2023-10-11:
We distribute tools via Microsoft's vcpgk (see https://artifacts.keil.arm.com/vcpkg-ce-registry/). In case of llvm embedded toolchain we refer directly to the release archives hosted on GitHub. 

Unfortunately, with the release 17.0.1 this doesn't work for the Mac distribution anymore. vcpkg does not support `.dmg` archives but only `.zip` or `.tar`. Would it be possible to provide a `.tar.xz` in addition to the `.dmg` for Mac?

JonatanAntoni, 2023-10-12:
Btw: Can we enhance the release workflow to register new releases automatically with the vcpkg registry? We could even decide to duplicate the release assets into the artifacts store if the network quota for GitHub hosted downloads becomes a bottleneck.

voltur01, 2023-10-12:
Hi Jonathan,

We do not have any automation in Github - the releases are created in downstream CI and uploaded manually following a release checklist.

I wonder if copying the packages into the Artifactory is a good idea anyway and we should change the approach now? Note that in this case you would probably have some downstream script to do this, where it should be possible to extract the `.dmg` e.g. see https://apple.stackexchange.com/questions/73926/is-there-a-command-to-install-a-dmg

What do you think?

JonatanAntoni, 2023-10-16:
@voltur01, it would be good if new releases would be uploaded into vcpkg right away. So, I'd propose to enhance the release checklist os that
- Adding vcpkg json file to artifactory, and
- Upload archives (optionally)
Would happen in as a normal step during release.

voltur01, 2024-01-18:
Internal process to publish into vcpkg was agreed, closing.

closed 2024-01-18 by voltur01