ARM-software/LLVM-embedded-toolchain-for-Arm #332: Could you add clang-tidy to the toolchain, please?

kmholtan, 2023-10-12:
This comes bundled with the releases of the llvm-project and had been convenient to get bundled here as well.

voltur01, 2023-10-12:
Hello,

LLVM Embedded Toolchain for Arm is based on exactly the same code as matching upstream releases, so advice is to make use of the upstream LLVM packages of matching version for any additional tools. They must be fully compatible.

We see rather limited value in recreating all the same packages that are already provided by upstream, on the contrary, the strategy is to get closer to upstream and, in the future, hopefully make embedded it part of the upstream.

closed 2023-10-12 by voltur01

kmholtan, 2023-10-12:
Thanks for the quick reply. Sorry for my limited knowledge of your toolchain, but how would such a merge work? With GNU you have the arm-none-eabi tools so you don’t get a conflict if you install gcc and the embedded toolchain at the same time. With LLVM-embedded all the executables, like clang share the same name as the upstream. Wouldn’t that be a challenge to merge, or are all the differences in libraries and none-executables? Could I replace the bin folder from your toolchain with the upstream (of the same version) and everything would work exactly the same?

voltur01, 2023-10-13:
Hi,

There could be two use cases, I think:
- Just for own use, one can install the upstream LLVM with required tools and LLVM ET (embedded toolchain) in a separate folder, then add LLVM ET `bin` path to `PATH` at the front, e.g. `export PATH=/.../LLVM-ET/bin:$PATH` - this way tools provided with LLVM ET will be selected before the upstream tools.
- Repackaging for redistribution, one can add additional tools from upstream LLVM package into the LLVM ET `bin` folder.