ARM-software/LLVM-embedded-toolchain-for-Arm #344: build from branch main looks broken

Laur59, 2023-10-31:
I tried to build from branch main (git hash 9b42be1) following the instructions :

```
mkdir build
cd build
cmake .. -GNinja -DFETCHCONTENT_QUIET=OFF
```

This was working for me in past (still works if I check out release 17.0.1) but now this ends up with a cmake error:

```
FAILED: llvmproject-populate-prefix/src/llvmproject-populate-stamp/llvmproject-populate-patch /Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/build/_deps/llvmproject-subbuild/llvmproject-populate-prefix/src/llvmproject-populate-stamp/llvmproject-populate-patch 
cd /Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/build/_deps/llvmproject-src && git reset --quiet --hard && git clean --quiet --force -dx && git am /Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/patches/llvm-project/0001-Fix-llvm-test-CodeGen-ARM-build-attributes.ll-test.patch /Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/patches/llvm-project/0002-Add-check-for-building-with-picolibc.patch /Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/patches/llvm-project/0003-Run-picolibc-tests-with-qemu.patch /Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/patches/llvm-project/0004-xfail-two-remaining-libcxx-with-picolibc-tests.patch && /Applications/CMake.app/Contents/bin/cmake -E touch /Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/build/_deps/llvmproject-subbuild/llvmproject-populate-prefix/src/llvmproject-populate-stamp/llvmproject-populate-patch
ninja: build stopped: subcommand failed.
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.27/Modules/FetchContent.cmake:1662 (message):
  Build step for llvmproject failed: 1
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.27/Modules/FetchContent.cmake:1802:EVAL:2 (__FetchContent_directPopulate)
  /Applications/CMake.app/Contents/share/cmake-3.27/Modules/FetchContent.cmake:1802 (cmake_language)
  /Applications/CMake.app/Contents/share/cmake-3.27/Modules/FetchContent.cmake:2016 (FetchContent_Populate)
  CMakeLists.txt:298 (FetchContent_MakeAvailable)
```

-- Configuring incomplete, errors occurred!

voltur01, 2023-10-31:
Hello,

Thank you for reporting the issue!

The error message, apparently, suggests that an internal error occurred in cmake itself. The log suggests that cmake 3.27 was used.

I have cmake 3.27.6 installed via brew on my macOS and the build completed successfully just a few minutes ago. Equally, the internal CI has successful builds for all platforms.

Would you please check if your cmake version can be updated?

Laur59, 2023-10-31:
Thank you for your fast reply !
I currently run version 3.27.7 of cmake, installed from the package available on cmake.org. I am running macOS  14.1 (Sonoma) on a MacBook Air M2 (2022).
I will then try to install cmake with brew and also try with release candidate of version 3.28.0

closed 2023-10-31 by Laur59