ARM-software/LLVM-embedded-toolchain-for-Arm #352: Can you support cross-compilation of ardupilot and px4?

Lqs66, 2023-11-09:
I'm looking for a way to cross compile open source flight controls using clang so I can do static analysis, does your tool support that?

voltur01, 2023-11-09:
Hello,

There seem to be two questions:
1. Cross compilation: Yes, this toolchain runs on a host environment, e.g. Linux or Windows, and compiles for an embedded Arm target.
2. Static analysis: This toolchain does not provide LLVM/clang tools not required for building, including tools like https://clang.llvm.org/docs/ClangStaticAnalyzer.html, to use these, please just install the matching version (17.x.x currently) from the main LLVM site https://releases.llvm.org/

Lqs66, 2023-11-09:
> Hello,
>
> There seem to be two questions:
>
> 1. Cross compilation: Yes, this toolchain runs on a host environment, e.g. Linux or Windows, and compiles for an embedded Arm target.
> 2. Static analysis: [...]

Thank you for your reply.
When I try to cross-compile ardupilot using this tool, the following problem occurs:

> ... /... /libraries/AP_Common/missing/cmath:3:15: fatal error: 'cmath' file not found
>     3 | #include_next <cmath>
>       | ^~~~~~~
> 1 warning and 1 error generated.

The compile command I entered is:
`CC=/home/lqs66/LLVMEmbeddedToolchainForArm-17.0.1-Linux-x86_64/bin/clang CXX=/home/lqs66/LLVMEmbeddedToolchainForArm- 17.0.1-Linux-x86_64/bin/clang++ CXXFLAGS="-O0 -g --target=arm-none-eabi -fno-rtti" . /waf configure --board=CUAVv5 && . /waf copter`

voltur01, 2023-11-09:
Hi,

I noticed that there is an extra space in the provided value for the `CXX` variable which may result in using a wrong C++ compiler.

More generally, building for an embedded target requires some additional command lines, please see https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm#using-the-toolchain

Note that when you use `--target=arm-none-eabi` the default Arm architecture is selected which is rather outdated, so you may want to be more specific, e.g. `--board=CUAVv5` seems to imply Cortex-M7 that is Armv7E-M.

Lqs66, 2023-11-09:
> Hi,
>
> I noticed that there is an extra space in the provided value for the `CXX` variable which may result in using a wrong C++ compiler.
>
> More generally, building for an embedded target requires [...]

Thank you very much for your reply!
It doesn't seem to be the cause of the problem, after changing --target=armV7e-m-none-eabi I still get this error, could you please try to compile it?

smithp35, 2023-11-09:
Looking at the ardupilot and px4 code-bases 

ardupilot in https://ardupilot.org/dev/docs/building-setup-linux.html says
```
The gcc-arm cross-compiler from [here](https://firmware.ardupilot.org/Tools/STM32-tools/) (ArduPilot is only built and tested on these specific versions of gcc-arm; if installed with apt-get gcc-arm will not produce a working binary in many cases)
```
px4 in https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html seems to be built on top of a linux targeting toolchain. 

Be warned that even if we solve this immediate problem there may be many others ahead. The best people to support a new toolchain is the open source project itself. We can fix problems with the toolchain, but there may be other incompatibilities that need changes to the open source project and we can't help with that.

With all that said. Can we first start with a simple example such as:
```
#include <cmath>

int main(void) {
    return 0;
}
```
With:
./LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/clang++ lib.cpp --target=arm-none-eabi -march=armv7-m -fno-rtti
 -fno-exceptions -c -v

With -v showing my include directories:

I can see:
 ./LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7m_soft_nofp/include/c++/v1
 ./LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/lib/clang/17/include
 ./LLVMEmbeddedToolchainForArm-17.0.0-Linux-x86_64/bin/../lib/clang-runtimes/arm-none-eabi/armv7m_soft_nofp/include
 
 We don't yet have a multilib entry for -march=arm7e-m, with that I get an error message for multilib not found:
```
clang++: warning: no multilib found matching flags: --target=thumbv7em-none-unknown-eabi -march=thumbv7em+nosha2+noaes+nofp16+nofp16fml -mfloat-abi=softfp -mfpu=fpv4-sp-d16 [-Wmissing-multilib]
clang++: note: available multilibs are:
--target=aarch64-none-unknown-elf
--target=armv4t-none-unknown-eabi -mfpu=none
--target=armv5e-none-unknown-eabi -mfpu=none
--target=thumbv6m-none-unknown-eabi -mfpu=none
--target=armv7-none-unknown-eabi -mfpu=none
--target=armv7-none-unknown-eabihf -mfpu=vfpv3-d16
--target=armv7r-none-unknown-eabi -mfpu=none
--target=armv7r-none-unknown-eabihf -mfpu=vfpv3-d16
--target=thumbv7m-none-unknown-eabi -mfpu=none
--target=thumbv7em-none-unknown-eabi -mfpu=none
--target=thumbv7em-none-unknown-eabihf -mfpu=fpv4-sp-d16
--target=thumbv7em-none-unknown-eabihf -mfpu=fpv5-d16
--target=thumbv8m.main-none-unknown-eabi -mfpu=none
--target=thumbv8m.main-none-unknown-eabihf -mfpu=fpv5-d16
--target=thumbv8.1m.main-none-unknown-eabi -mfpu=none
--target=thumbv8.1m.main-none-unknown-eabihf -march=thumbv8.1m.main+fp16 -mfpu=fp-armv8-fullfp16-sp-d16
--target=thumbv8.1m.main-none-unknown-eabihf -march=thumbv8.1m.main+dsp+mve -mfpu=none
```

I know we did have a problem with C++ header directories when more than one target satisfied the multilib. The symptoms where several C++ header paths "include/c++/v1" followed by clang/17/include (runtimes) followed by several C header paths include.

If you can get a simple example working then try the configure step using these options.

Lqs66, 2023-11-09:
> Thank you very much for your reply! It doesn't seem to be the cause of the problem, after changing --target=armV7e-m-none-eabi I still get this error, could you please try to compile it?

Thanks for the reply, I think I got it.

voltur01, 2024-01-18:
Looks like the questions were answered, closing.

closed 2024-01-18 by voltur01