ARM-software/LLVM-embedded-toolchain-for-Arm #365: Linker select strange runtime library

Laur59, 2023-11-27:
With toolchain built from branch main ID 2e4582ce I try to build a system from C file sources. The C files are compiled with for instance for kern.c

```
clang --target=thumbv7em-none-unknown-eabihf \
-c \
-o kern.o \
-mcpu=cortex-m4 \
-mfloat-abi=hard -mfpu=fpv4-sp-d16 \
-fno-exceptions -fno-rtti \
kern.c
```

Then one use the following command to link:

```
clang --target=thumbv7em-none-unknown-eabihf -v \
-L/Users/lva/Dev/u111/kernel-os/Ports/EquatesModels/Cores/CORTEX_M4/Runtime \
-mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fno-exceptions -fno-rtti \
-T/Users/lva/Dev/u111/kernel-os/Ports/Targets/Nucleo_4R5/Base/Runtime/link.ld \
-Wl,-Map=EPROM.map \
	first.o \
	 librtcb.a liblibx.a libproc.a libtool.a libkern.a   \
	-lm -o EPROM.nosig.elf
```

The above command prints on stdout:

```
clang version 18.0.0
Target: thumbv7em-none-unknown-eabihf
Thread model: posix
InstalledDir: /Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/build/llvm/bin
 "/Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/build/llvm/bin/ld.lld"
 -Map=EPROM.map first.o librtcb.a liblibx.a libproc.a libtool.a libkern.a -lm -Bstatic -EL
 -L/Users/lva/Dev/u111/kernel-os/Ports/EquatesModels/Cores/CORTEX_M4/Runtime
 -T /Users/lva/Dev/u111/kernel-os/Ports/Targets/Nucleo_4R5/Base/Runtime/link.ld
 -L/Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/build/llvm/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv4_sp_d16_exceptions_rtti/lib
 -L/Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/build/llvm/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv4_sp_d16_exceptions_rtti/lib
 -L/Users/scratch/LLVM-ARM/LLVM-embedded-toolchain-for-Arm-macOS/build/llvm/bin/../lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv4_sp_d16_exceptions_rtti/lib
 -lc -lm -lclang_rt.builtins --target2=rel -o EPROM.nosig.elf
```

I do no understand why the linker is using the runtime library `lib/clang-runtimes/arm-none-eabi/armv7em_hard_fpv4_sp_d16_exceptions_rtti/lib` if we use the options `-fno-exceptions -fno-rtti`

voltur01, 2024-01-18:
Upstream PR https://github.com/llvm/llvm-project/pull/75031

Laur59, 2024-02-29:
This issue can be closed, the toolchain built with ID 0e869035 (and with ID 9c7cde64 of the upstream) use the libraries according to flags -fno-exceptions -fno-rtti

Laur59, 2024-02-29:
This issue can be closed, the toolchain built with ID https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/commit/0e8690358532635468e16d4a8238e561493dff17 (and with ID 9c7cde64 of the upstream) use the libraries according to flags -fno-exceptions -fno-rtti

closed 2024-02-29 by Laur59