ARM-software/LLVM-embedded-toolchain-for-Arm #376: In the -Oz optimization mode, when enabling -mbranch-protection=standard, the program encounters an error.

zhou-shan, 2023-12-20:
1. The code that caused the error is as follows:(test.c)

`struct i3c_dev_desc;

struct i3c_master_controller_ops {
  int (*f1)(void);
  int (*f2)(void);
  int (*f3)(void);
  int (*f4)(void);
  int (*f5)(void);
  int (*f6)(void);
  int (*f7)(void);
  int (*f8)(void);
  int (*priv_xfers)(struct i3c_dev_desc *dev, char *xfers, int nxfers);
};

struct i3c_master_controller {
  int a;
  int b;
  int c;
  unsigned long d;

  struct i3c_master_controller_ops *ops;
};

struct i3c_i2c_dev_desc {
  char *a;
  char *b;
  struct i3c_master_controller *master;
  void *master_priv;
};

struct i3c_dev_desc {
  struct i3c_i2c_dev_desc common;
};

#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wincompatible-pointer-types"
int i3c_dev_do_priv_xfers_locked(struct i3c_dev_desc *dev,
         char *xfers,
         int nxfers)
{
  struct i3c_master_controller *master;

  if (!dev)
    {
      return -2;
    }

  master = dev->common.master;
  if (!master || !xfers)
    {
      return -22;
    }

  if (!master->ops->priv_xfers)
    {
      return -138;
    }

  return master->ops->priv_xfers(dev, xfers, nxfers);
}
#pragma GCC diagnostic pop`

2. Compile options using clang 15.0.2 or 17.0.1 are as follows:
clang -c --target=armv8.1-m.main -mcpu=cortex-m85 -mfloat-abi=soft -mthumb -mpure-code -mbranch-protection=standard -Wall -Oz -fno-builtin -fshort-enums -fno-common -fsigned-char -fomit-frame-pointer -ffunction-sections -fdata-sections -fstack-protector-strong -fno-unroll-loops -std=c99 -g test.c -o test.o
3. Problem Description:
[image]
When executing "bx r12" to jump to "master->ops->priv_xfers", the value of r12 (which holds the PAC) was modified due to PAC verification earlier, causing an unexpected behavior of "bx r12" instruction and the program to run out of control. (In fact, PAC verification was not necessary before the "bx" instruction, considering the semantics of this function.)

zhou-shan, 2023-12-20:
This should be a compiler bug.

Experimental conclusion: Only when all of the following four conditions are met, will similar problems occur.
- Enable -mbranch-protection=standard: Enable the compiler's branch protection (PAC/BTI) instrumentation feature.
- Set optimization level to -Oz.
- Enable -foptimize-sibling-calls (enabled by default under -O2/-O3/-Os/-Oz): Optimize sibling and tail recursive calls. This reduces the overhead of function calls, such as tail call functions.
- The last statement of the function exit is in the form of "return a function pointer", such as return func(...);

The current temporary solution is:
- Option 1: If you want to use the -Oz optimization level, it is recommended to add -fno-optimize-sibling-calls to eliminate the compiler optimization bug in these corner & cross situations.
- Option 2: Use the -Os optimization level.

stuij, 2024-01-15:
This is the exact same issue as https://github.com/llvm/llvm-project/issues/75998. I don't think there's a point in keeping this duplicate ticket around. This is not an issue specifically related to the embedded toolchain. (and we're already tracking the LLVM ticket)

closed 2024-01-15 by stuij