ARM-software/LLVM-embedded-toolchain-for-Arm #393: Could we have support for _RETARGETABLE_LOCKING also for newlib?

piotrrak, 2024-02-28:
Hello,

I am working on adding support of cortex-m85 in ZephyrOS.
One of my bullets is supporting LLVM-embedded-toolchain-for-Arm 
Also fixing support of cortex-m55 for since it never worked for clang + armv8.1m mix in ZephyrOS.

Multithreading in ZephyrOS requires _RETARGETABLE_LOCKING which is currently enable for picolibc bundled with LLVM-embedded-toolchain-for-ARM, but disabled for newlib overlay.
It seems gcc based toolchain also enables _RETARGETABLE_LOCKING enables this feature.

Although I didn't check other releases than other releases than 18_devdrop_0 it is not enabled for any of newlib runtimes at least in this release.

generated
`lib/clang-runtimes/newlib/*/*/newlib.h`
currently has: 
```
/* Define if using retargetable functions for default lock routines.  */
/* #undef _RETARGETABLE_LOCKING */
```
Cheers,
/P

voltur01, 2024-02-28:
Hi Piotr,

Thank you for reporting this, we will look into enabling _RETARGETABLE_LOCKING in the newlib package for the upcoming LLVM 18 based release.

piotrrak, 2024-02-28:
That's great! Thank you.

voltur01, 2024-03-15:
Done in https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/400