ARM-software/LLVM-embedded-toolchain-for-Arm #437: Unclear how to do a build including runtime libraries from source on windows

klaus1212, 2024-05-01:
Hi,

I am trying to build a clang compiler with windows 64bit as the host system and arm as the target system. But it is NOT clear to me from the "building-from-source.md" how to actually build both the host system compiler and the target runtime libraries on windows.

building-from-source.md seems to be written pretty much with ubuntu as the host system in mind. It is less clear how to actually do the building using windows as the host system.
The section "Host platforms" says:
- LLVM Embedded Toolchain for Arm is built and tested on Ubuntu 18.04 LTS.
- The Windows version is built on Windows Server 2019 and lightly tested on Windows 10.

Since I am trying to build on windows, I read above as that should be possible. To do so I have more or less done what building-from-source.md suggests in the "Building" section. However it fails in a configure step of newlib. Basically because configure for newlib is a bash script, and cmake on windows  generates a build system that calls configure from a windows cmd prompt which in turn fails.

To me it looks like it is not possible to build LLVM-embedded-toolchain-for-Arm on windows including the arm specific runtime libraries. Am I wrong or am I missing something, please advise or be more clear about this in the building from source.

For reference here is what I have done in detail to build on windows:
I am using the 18.1.3 release of LLVM and LLVM-embedded-toolchain-for-Arm. Additionally I use newlib-4.3.0.

My runtime libraries needs to be for -marm and -mthumb:
add_library_variants_for_cpu(
    armv7a
    SUFFIX hard_neon
    COMPILE_FLAGS "-mfloat-abi=hard -march=armv7a -mfpu=neon"
    MULTILIB_FLAGS "--target=armv7-none-unknown-eabihf -mfpu=neon"
    QEMU_MACHINE "none"
    QEMU_CPU "cortex-a8"
    QEMU_PARAMS "-m 1G"
    BOOT_FLASH_ADDRESS 0x00000000
    BOOT_FLASH_SIZE 0x1000
    FLASH_ADDRESS 0x20000000
    FLASH_SIZE 0x1000000
    RAM_ADDRESS 0x21000000
    RAM_SIZE 0x1000000
    STACK_SIZE 4K
)
add_library_variants_for_cpu(
    armv7a
    SUFFIX thumb_hard_neon
    COMPILE_FLAGS "-mfloat-abi=hard -march=armv7a -mfpu=neon"
    MULTILIB_FLAGS "--target=thumbv7-none-unknown-eabihf -mfpu=neon"
    QEMU_MACHINE "none"
    QEMU_CPU "cortex-a8"
    QEMU_PARAMS "-m 1G"
    BOOT_FLASH_ADDRESS 0x00000000
    BOOT_FLASH_SIZE 0x1000
    FLASH_ADDRESS 0x20000000
    FLASH_SIZE 0x1000000
    RAM_ADDRESS 0x21000000
    RAM_SIZE 0x1000000
    STACK_SIZE 4K
)

I have then modified LLVM-embedded-toolchain-for-Arm\CMakeLists.txt with the following:
option(
    LLVM_TOOLCHAIN_CROSS_BUILD_MINGW
    "Cross-build for Windows. Using this option implies that you accept the GCC & MinGW licenses."
    OFF
)
option(
    PREBUILT_TARGET_LIBRARIES
    "Target libraries are prebuilt so no need to build them"
    OFF
)
set(LLVM_TOOLCHAIN_LIBRARY_VARIANTS
    "armv7a_hard_neon_exn_rtti;armv7a_thumb_hard_neon_exn_rtti" CACHE STRING
    "Build only the specified library variants. If not specified then build all variants."
)
set(LLVM_TOOLCHAIN_C_LIBRARY
    "newlib" CACHE STRING
    "Which C library to use."
)
For further details on my modifications to the cmakelists.txt of LLVM-embedded-toolchain-for-Arm I have attached a LLVM-embedded-toolchain-for-Arm_Windows.patch

Then I create the cmake project using ninja from a "X64 native tools command prompt for vs2022" like:
cd C:\code\LLVM-embedded-toolchain-for-Arm\build\vs2022
cmake ../.. -G Ninja -DFETCHCONTENT_QUIET=OFF -DFETCHCONTENT_SOURCE_DIR_LLVMPROJECT=../../../llvm-project -DFETCHCONTENT_SOURCE_DIR_NEWLIB=../../../newlib-cygwin

and get a ninja build system using the visual studio compiler.

This compiles fine until I get to the target libraries.
These fail in the configure step for newlib, basically because configure for newlib is a bash script, and cmake on windows  generates a build system that calls configure from a windows cmd prompt which in turn fails. I have attached the build prompt for reference.
The error for the configure step looks like:
%1 is not a valid Win32 application.

[x64 Native Tools Command Prompt for VS 2022.txt](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/files/15175094/x64.Native.Tools.Command.Prompt.for.VS.2022.txt)

[LLVM-embedded-toolchain-for-Arm_Windows.patch](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/files/15175081/LLVM-embedded-toolchain-for-Arm_Windows.patch)

voltur01, 2024-05-01:
Hi,

Thank you for raising this issue and providing the details!

Unfortunately, you are correct that newlib does not support building on Windows properly - this was one of the key points why we switched to picolibc in later releases.

Up to and including version 14, newlib was the default library and the Windows building instructions advised to use MinGW, see https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/llvm-14/docs/building-from-source.md and corresponding package https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases/tag/release-14.0.0

It should be possible to adapt the instructions to work with the latest release, however as https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/docs/newlib.md says newlib support is still considered to be experimental because of the issues like Windows building, clang incompatibilities that need to be patched and so on.

I will update the docs to highlight that newlib is now build on Linux, indeed.

klaus1212, 2024-05-01:
Hi,

Thank you for the quick reply, looking forward to the highlighting that newlib needs to be build on windows :-)

Since you say it is related to newlib, does this mean that I should be able to build the runtimes too, with windows 64bit as the host system and arm as the target system, if I had used piclibc instead?

voltur01, 2024-05-01:
Hi,

Upon double-checking I have found the following:
- In our CI we build LLVM on Windows natively, but re-use runtimes and libraries built on Linux.
- picolibc refers to mingw for Windows builds as well https://github.com/picolibc/picolibc

So the full build on Windows seems to require mingw even for picolibc.

voltur01, 2024-05-02:
Docs clarified in https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/439, closing.

closed 2024-05-02 by voltur01