ARM-software/LLVM-embedded-toolchain-for-Arm #448: Can debug info be removed from lib*.a files ?

Laur59, 2024-06-17:
Hi!
When trying to build a disassembly file from an elf file one get warnings about source files not found. For instance

```
/Volumes/LLVM-ET-Arm-18.1.3-Darwin-universal/LLVM-ET-Arm-18.1.3-Darwin-universal/bin/llvm-objdump -S FLASH.elf > FLASH.dis
/Volumes/LLVM-ET-Arm-18.1.3-Darwin-universal/LLVM-ET-Arm-18.1.3-Darwin-universal/bin/llvm-objdump: warning: 'FLASH.elf': failed to find source /workspace/workspace/llvm-embedded/llvm-18/build-and-test/build/picolibc/armv7em_hard_fpv4_sp_d16_exn_rtti/src/picolibc_armv7em_hard_fpv4_sp_d16_exn_rtti-build/../../../../../src/picolibc/newlib/libc/ssp/stack_protector.c
.
.
.
```

I tried modifying CMakeLists.txt to replace 

PICOLIBC_BUILD_TYPE "minsize"

by

PICOLIBC_BUILD_TYPE "release"

as from my understanding of meson buildtype, minsize keeps debug whereas release does not.

But this was not solving the issue.

For now the workaround is to strip the installed libraries.

```
find ${PREFIX}/lib/clang-runtimes/arm-none-eabi -name "lib*.a" -exec ${PREFIX}/bin/llvm-strip --strip-unneeded {} \;
```

Any hint on how to modify CMakeLists.txt or options to be passed when invoking cmake is welcome. Thank you !

voltur01, 2024-06-17:
Hi,

Looking at the docs, https://mesonbuild.com/Builtin-options.html says that `debug` is there to "Enable debug symbols and other information."

So presumably `debug=false` meson option would remove debug symbols, however the table for `buildtype` https://mesonbuild.com/Builtin-options.html#details-for-buildtype already says that it implies `debug=false` for the `release` option, so changing the build type to release, as you suggested, should have worked.

May be a question for https://github.com/mesonbuild/meson

Laur59, 2024-06-23:
Seems that changing also in CMakeLists.txt 

-DCMAKE_BUILD_TYPE=MinSizeRel

by

-DCMAKE_BUILD_TYPE=Release

is solving my question.

closed 2024-06-23 by Laur59