ARM-software/LLVM-embedded-toolchain-for-Arm #474: PicoLIbc for Aarch64 gives missing limits.h header build error

itxs, 2024-08-30:
While doing the build according to building-from-source.md, I cloned repos and applied all mentioned patches, started building using Ninja, and then I faced an issue related to PicoLibc and aarch64, it looks like, PicoLibc, which is selected by default, needs additional configuration definitions to be used under aarch64, or requires additional patches. However, I need only the ARM target architecture.

```
  142 | # include_next <limits.h>
      |                ^~~~~~~~~~
1 error generated.
[14/282] Building C object lib/builtin....builtins-aarch64.dir/comparetf2.c.obj
FAILED: lib/builtins/CMakeFiles/clang_rt.builtins-aarch64.dir/comparetf2.c.obj 
/home/itxs/LLVM-embedded-toolchain-for-Arm/build/llvm/bin/clang --target=aarch64-none-elf -DHAS_ASM_LSE -DVISIBILITY_HIDDEN  --target=aarch64-none-elf -march=armv8-a -ffunction-sections -fdata-sections -fno-ident --sysroot /home/itxs/LLVM-embedded-toolchain-for-Arm/build/llvm/lib/clang-runtimes/aarch64-none-elf/aarch64_exn_rtti -Wall -Wno-unused-parameter -O3 -DNDEBUG -fno-lto -Werror=array-bounds -Werror=uninitialized -Werror=shadow -Werror=empty-body -Werror=sizeof-pointer-memaccess -Werror=sizeof-array-argument -Werror=suspicious-memaccess -Werror=builtin-memcpy-chk-size -Werror=array-bounds-pointer-arithmetic -Werror=return-stack-address -Werror=sizeof-array-decay -Werror=format-insufficient-args -Wformat -std=c11 -fPIC -fno-builtin -fvisibility=hidden -fomit-frame-pointer -DCOMPILER_RT_HAS_FLOAT16 -MD -MT lib/builtins/CMakeFiles/clang_rt.builtins-aarch64.dir/comparetf2.c.obj -MF lib/builtins/CMakeFiles/clang_rt.builtins-aarch64.dir/comparetf2.c.obj.d -o lib/builtins/CMakeFiles/clang_rt.builtins-aarch64.dir/comparetf2.c.obj -c /home/itxs/LLVM-embedded-toolchain-for-Arm/repos/llvm-project/compiler-rt/lib/builtins/comparetf2.c
In file included from /home/itxs/LLVM-embedded-toolchain-for-Arm/repos/llvm-project/compiler-rt/lib/builtins/comparetf2.c:40:
In file included from /home/itxs/LLVM-embedded-toolchain-for-Arm/repos/llvm-project/compiler-rt/lib/builtins/fp_lib.h:23:
In file included from /home/itxs/LLVM-embedded-toolchain-for-Arm/repos/llvm-project/compiler-rt/lib/builtins/int_lib.h:93:
In file included from /home/itxs/LLVM-embedded-toolchain-for-Arm/build/llvm/lib/clang/20/include/limits.h:25:
/home/itxs/LLVM-embedded-toolchain-for-Arm/build/llvm/lib/clang-runtimes/aarch64-none-elf/aarch64_exn_rtti/include/limits.h:142:16: fatal error: 'limits.h' file not found
  142 | # include_next <limits.h>
      |                ^~~~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
[5/708] Performing install step for 'picolibc_aarch64'
FAILED: compiler-rt/aarch64_exn_rtti/src/compiler_rt_aarch64_exn_rtti-stamp/compiler_rt_aarch64_exn_rtti-build /home/itxs/LLVM-embedded-toolchain-for-Arm/build/compiler-rt/aarch64_exn_rtti/src/compiler_rt_aarch64_exn_rtti-stamp/compiler_rt_aarch64_exn_rtti-build 
cd /home/itxs/LLVM-embedded-toolchain-for-Arm/build/compiler-rt/aarch64_exn_rtti/src/compiler_rt_aarch64_exn_rtti-build && /usr/bin/cmake --build .
[7/708] Performing install step for 'picolibc_aarch64'
ninja: Entering directory `/home/itxs/LLVM-embedded-toolchain-for-Arm/build/picolibc/aarch64/src/picolibc_aarch64-build'
ninja: no work to do.
ninja: build stopped: subcommand failed.
```

vrukesh, 2024-08-30:
Hi Timur,
Hope you have pulled latest picolibc code, this issue has been fixed a couple of days earlier with: https://github.com/picolibc/picolibc/pull/810

After this fix, I feel you may also observe another unrelated issue of duplicate symbols. This also can be fixed by temporarily using older picolibc version: https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/465

Regards,
Vrukesh

itxs, 2024-09-06:
UPD: Thanks, I updated my picolibc and the issue gone