ARM-software/LLVM-embedded-toolchain-for-Arm #475: error: Failed to merge in the changes

AspenBen, 2024-08-31:
I tried to compile the llvm myself,  so I download the code and follow the [guide](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/docs/building-from-source.md)

but it occurred error that refer to the picture log, I have no idea what to do next.
[image]

I believe I have installed the necessary tool to compile.

voltur01, 2024-09-02:
Hi, it looks like there is a merge conflict because of recent upstream changes, seems to be the same as this https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues/476 - we will look into updating this ASAP.

statham-arm, 2024-09-02:
Believed fixed by #477.

closed 2024-09-02 by statham-arm