ARM-software/LLVM-embedded-toolchain-for-Arm #488: Problem linking address sanitizer

arifbalik, 2024-09-05:
I'm passing `-fsanitize=address` flag to my compiler and linker to debug my project. However I'm unable to link the project because the linker can not find the symbols for asan

This is my link command;

```bash
clang \
-target armv6m-none-eabi \
-mfpu=none \
-ffreestanding \
-fno-builtin \
-fno-common\
-ffunction-sections \
-fdata-sections \
-O0 \
-g \

-fsanitize=address \

-fuse-ld=lld \
-Wl,--gc-sections \
-T stm32f030x8.ld \
CMakeFiles/eup4520.dir/src/main.c.obj \
CMakeFiles/eup4520.dir/src/eu44020tk_stdkey_4k2ld.c.obj \
CMakeFiles/eup4520.dir/src/eup4420_variables_visibility_settings.c.obj \
-o eup4520.elf  \
3rdparty/lib3rdparty.a \
lib/hal/libhal.a  \
lib/thermostat/libthermostat.a \
lib/ui/libui.a  \
lib/nfc/libnfc.a
```

But again I get linking errors like;

`ld.lld: error: undefined symbol: __asan_report_load2`

Any help would be welcome

voltur01, 2024-09-05:
Hi Arif, unfortunately, Address Sanitizer and Memory Sanitizer do not support bare-metal configurations yet and they require a big redesign to make them work for microcontrollers, thus support is not expected any time soon.

arifbalik, 2024-09-05:
I see, i managed to make it work with mocking on x86 but never tried on bare-metal. Thats too bad :/