ARM-software/LLVM-embedded-toolchain-for-Arm #492: Release 18.1.3 fails to build (duplicate names in newlib/libc.a)

begriffs, 2024-09-07:
Fails similarly on the main branch ([6d76d5](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/commit/6d76d5c9b8ae294409c8e147ea1c3e3281cc2d1d)).

System information:
* OpenBSD build 7.5 GENERIC.MP#82 amd64
* python 3.11.9
* cmake 3.28.3
* ninja 1.11.1
* clang 16.0.6
* meson 1.3.2

Steps to reproduce:

```sh
cd ~/dev
git clone https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm.git
cd LLVM-embedded-toolchain-for-Arm
git checkout release-18.1.3

export CC=clang
export CXX=clang++
mkdir build
cd build
cmake .. -GNinja -DFETCHCONTENT_QUIET=OFF

ninja llvm-toolchain
```

The crux of the error appears to be this part of the output:

```
FAILED: newlib/libc_duplicates
~/dev/LLVM-embedded-toolchain-for-Arm/build/_deps/picolibc-src/scripts/duplicate-names /usr/bin/nm newlib/libc.a newlib/libc_duplicates
Duplicate names in newlib/libc.a
00000000 F k_rem_pio2.c
00000000 F s_scalbln.c
```

begriffs, 2024-09-08:
cc @keith-packard since it's happening in the picolibc build

keith-packard, 2024-09-08:
Looks like my duplicate name checking code is busted -- it's finding some duplicate *file* names, not duplicate *function* names and complaining about that.

This patch might help?
```
diff --git a/scripts/duplicate-names b/scripts/duplicate-names
index ace926157..b14741f52 100755
--- a/scripts/duplicate-names
+++ b/scripts/duplicate-names
@@ -3,7 +3,7 @@ NM="$1"
 FILE="$2"
 OUTPUT="$3"
 
-"$NM" -g "$FILE" 2>/dev/null | grep ' [A-TVX-Z] ' | grep -v '__x86' | sort | uniq -d > "$OUTPUT"
+"$NM" -g "$FILE" 2>/dev/null | grep ' [A-EG-TVX-Z] ' | grep -v '__x86' | sort | uniq -d > "$OUTPUT"
 if [ -e "$OUTPUT" ] && [ -s "$OUTPUT" ]; then
     echo "Duplicate names in ${FILE}"
     cat "$OUTPUT"
```

begriffs, 2024-09-09:
Wow fast response Keith, thanks! That patch seems to fix the error message I encountered before. However, `ninja check-llvm-toolchain` still fails and I don't know why.

I created a fresh OpenBSD 7.5 server and captured full logs of the build which are attached to this comment.

* cmake.log
   * the output of `cmake .. -GNinja -DFETCHCONTENT_QUIET=OFF`
* cmake2.log
   * the first time I ran the above command, my git name/email were not configured which caused an error. I set those and ran the command again.
* ninja.log.gz
   * the output of `ninja llvm-toolchain`
* check.log
   * the output of `ninja check-llvm-toolchain`

Hope someone can spot the problem. Last summer (IIRC) this project did build successfully for me on OpenBSD 7.3.

---

[ninja.log.gz](https://github.com/user-attachments/files/16930816/ninja.log.gz)
[cmake2.log](https://github.com/user-attachments/files/16930817/cmake2.log)
[cmake.log](https://github.com/user-attachments/files/16930818/cmake.log)
[check.log](https://github.com/user-attachments/files/16930819/check.log)

keith-packard, 2024-09-09:
The patch I showed is already upstream, and I'll be releasing that shortly. Thanks for testing!

begriffs, 2024-09-24:
I started over on a fresh VM and got commit bd5ecae8e6ad517410fed0143d95fedaf423e2c7 to build successfully.

However, `ninja check-llvm-toolchain` fails. Here are the relevant errors at the end of the output:
```
FAILED: test/test-cplusplus.p/test-cplusplus.cpp.o 
clang++ -Itest/test-cplusplus.p -Itest -I../../../../_deps/picolibc-src/test -Inewlib/libm/common -I../../../../_deps/picolibc-src/newlib/libm/common -Inewlib/libc/machine/aarch64 -I../../../../_deps/picolibc-src/newlib/libc/machine/aarch64 -Inewlib/libc/tinystdio -I../../../../_deps/picolibc-src/newlib/libc/tinystdio -I. -I../../../../_deps/picolibc-src -Inewlib/libc/include -I../../../../_deps/picolibc-src/newlib/libc/include -Isemihost -I../../../../_deps/picolibc-src/semihost -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -O3 -g -nostdlib -fno-common -frounding-math -Wno-unsupported-floating-point-opt -fno-builtin-copysignl -Werror=vla -Warray-bounds -Werror=double-promotion -Wno-missing-braces -Wno-return-type -Wno-unused-command-line-argument -Wmissing-prototypes -Wmissing-declarations -fno-exceptions -fno-unwind-tables -fno-stack-protector -D_FORTIFY_SOURCE=3 -MD -MQ test/test-cplusplus.p/test-cplusplus.cpp.o -MF test/test-cplusplus.p/test-cplusplus.cpp.o.d -o test/test-cplusplus.p/test-cplusplus.cpp.o -c ../../../../_deps/picolibc-src/test/test-cplusplus.cpp
In file included from ../../../../_deps/picolibc-src/test/test-cplusplus.cpp:44:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/arpa/inet.h:5:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/endian.h:40:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/machine/endian.h:5:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/sys/cdefs.h:45:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/sys/config.h:34:
In file included from /usr/include/c++/v1/float.h:73:
In file included from /usr/include/c++/v1/__config:313:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/sys/endian.h:33:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/sys/_types.h:52:
In file included from /usr/include/c++/v1/stddef.h:17:
/usr/include/stddef.h:44:9: error: unknown type name '__ptrdiff_t'
typedef __ptrdiff_t     ptrdiff_t;
        ^
/usr/include/stddef.h:49:9: error: unknown type name '__size_t'
typedef __size_t        size_t;
        ^
/usr/include/stddef.h:60:9: error: unknown type name '__wint_t'
typedef __wint_t        wint_t;
        ^
/usr/include/stddef.h:65:9: error: unknown type name '__mbstate_t'
typedef __mbstate_t     mbstate_t;
        ^
/usr/include/stddef.h:78:26: error: expected ';' at end of declaration list
        long long __max_align_ll __aligned(__alignof__(long long));
                                ^
/usr/include/stddef.h:79:28: error: expected ';' at end of declaration list
        long double __max_align_ld __aligned(__alignof__(long double));
                                  ^
In file included from ../../../../_deps/picolibc-src/test/test-cplusplus.cpp:44:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/arpa/inet.h:5:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/endian.h:40:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/machine/endian.h:5:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/sys/cdefs.h:45:
In file included from ../../../../_deps/picolibc-src/newlib/libc/include/sys/config.h:34:
In file included from /usr/include/c++/v1/float.h:80:
/usr/include/float.h:38:1: error: unknown type name '__BEGIN_DECLS'
__BEGIN_DECLS
^
```
There are other warnings later, but this part seems to be the worst of it.

keith-packard, 2024-09-24:
That looks like it's trying to build the picolibc C++ header test. Maybe that just doesn't work in this environment? Are you expecting the toolchain to have C++ support?

begriffs, 2024-09-25:
Not sure about other people's needs, but I don't personally use C++.