ARM-software/LLVM-embedded-toolchain-for-Arm #510: Build lldb as well

begriffs, 2024-09-25:
My use case is to debug a micro-controller using lldb through OpenOCD's gdbserver interface. My system lldb is a lower version than the clang that LLVM-embedded-toolchain-for-Arm builds. Also, lldb probably needs to be [built with support](https://lldb.llvm.org/resources/build.html#cross-compiling-lldb) for the remote architecture.

Is it possible to make a new `ninja` target in this project to build an lldb with support for the same targets that its clang supports?

voltur01, 2024-10-10:
Hi Joe,

The readme suggests:

> Binary releases of the LLVM Embedded Toolchain for Arm are based on release branches of the upstream LLVM Project, thus can safely be used with all tools provided by LLVM [releases](https://github.com/llvm/llvm-project/releases) of matching version.

We believe it is easier to use LLVM binaries, otherwise effectively rebuilding `lldb` from the embedded toolchain source would be the same as rebuilding it from LLVM original source, from the same branch.

`lldb` (the client) includes support for all architectures, the link above refers to the need to rebuild the _server_ component to be able to run a remote _host_ - which is not possible for a bare-metal target. The [samples readme](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/tree/main/samples#compiling-running-and-debugging-the-samples) provides a couple of commands to make use of `lldb` with QEMU, should be similar with OpenOCD.

begriffs, 2024-10-12:
Thanks for explaining how it works. Didn't know that a plain install of lldb understands all architectures. I confirmed that my system lldb can load my binary. Target list shows

```
target #0: /path/to/my_image.dbg ( arch=arm-*-*-eabihf, platform=remote-openbsd )
```

Having trouble getting it to work sensibly with openocd, but that's probably unrelated to lldb's build options, so I'm closing this issue.

closed 2024-10-12 by begriffs