ARM-software/LLVM-embedded-toolchain-for-Arm #524: iostream is not available with newlib

nolange, 2024-10-07:
newlib libc++ is compiled with `_LIBCPP_HAS_NO_LOCALIZATION`, this means that iostreams are not available at all.

I am not asking for fully supporting the locales, but provide simple C-only locale stubs.

the `picolib` variant does this, and `newlib` should be the more complete variant?

smithp35, 2024-10-07:
Thanks for pointing that out. Our newlib overlay package is still in the experimental state at the moment. We're hoping to improve that in the future, particularly when newlib support for building with clang without patches is available.

It isn't currently a priority so this may be a case of "patches welcome" until we can get to it.

voltur01, 2024-10-07:
Hi, 

Arm Compiler for Embedded is a commercial toolchain that uses `armclang` compiler that is based on `clang`, but the rest of the toolchain components are proprietary.

This project - LLVM Embedded Toolchain for Arm - is an attempt to create a working and easy to use embedded toolchain using open-source components only.

So these are two very different toolchains based on the same great LLVM technology at the core.

closed 2025-01-23 by voltur01