ARM-software/LLVM-embedded-toolchain-for-Arm #539: [LLD][ARM] Link fails when using -frwpi and debug symbols

harrychin, 2024-10-22:
I have been using this toolchain for cross-platform builds on a Darwin host with great success, thank you, but I've been attempting to build for a RWPI thumbv8m.main-none-eabi target and lld always fails on "has non-ABS relocation R_ARM_SBREL32 against symbol".

There is an issue on upstream LLVM which seems to be the [same issue](https://github.com/llvm/llvm-project/issues/89197).

Can be reproduced from Clang 19 to 14.
https://godbolt.org/z/osKjoshjG

voltur01, 2024-10-23:
Hi, thank you for using the toolchain and reporting this issue - it is on our radar and we plan to look into this for LLVM 20 release.

ostannard, 2024-12-11:
Fixed by https://github.com/llvm/llvm-project/pull/116956

closed 2024-12-11 by ostannard