ARM-software/LLVM-embedded-toolchain-for-Arm #556: Provide libnosys.a for newlib compatibility

nolange, 2024-11-05:
Many projects use `GROUP(-lnosys)` in their ldscript, this allows to lazily pull in some no-op stubs.
Taking it out will cause trouble if newlib still should be usable.
(picolib already has `libg.a` for probably similar reasons)

this is what I had to do to fix the issue for me:

```
#!/bin/sh
PREFIX=${1:-/opt/arm-clang}
TEMPFILE=$(mktemp); trap "rm -f $TEMPFILE" 0

"${PREFIX}"/bin/llvm-ar qc $TEMPFILE && "${PREFIX}"/bin/llvm-ranlib $TEMPFILE

for d in "${PREFIX}"/lib/clang-runtimes/arm-none-eabi/*/lib; do
	install -m644 $TEMPFILE "$d"/libnosys.a
done
```

voltur01, 2024-11-05:
Hi, the `libg.a` example suggests that this belongs to https://github.com/picolibc/picolibc project itself - would you like to raise it as an issue there to discuss?

nolange, 2024-11-05:
Ok: https://github.com/picolibc/picolibc/issues/876

closed 2024-12-16 by voltur01