ARM-software/LLVM-embedded-toolchain-for-Arm #566: Undeclared identifier 'NULL'

Pantau123, 2024-11-14:
Hi,
i have a sample Program generated with the stcubemx, original gcc support. 
With the arm-gcc all compiles fine. 
I want to switch to LLVM-embedded-toolchain-for-ARM. I have installed the latest version, and i small "HelloWorld.c" compiles fine.
Now i want to compile my STCubeMX Program and i receive the following errors:
  FAILED: CMakeFiles/TimerTest_Nucleo_F303-RE.dir/Core/Src/sysmem.c.obj 

> LLVM-ET-Arm-19.1.1-Windows-x86_64\bin\clang.exe --sysroot=F:/test/LLVM-ET-Arm-19.1.1-Windows-x86_64/lib/clang-runtimes/arm-none-eabi/armv7m_hard_fpv4_sp_d16 -DDEBUG -DSTM32F303xE -DUSE_HAL_DRIVER -IF:/test/LLVM-ET-Arm-19.1.1-Windows-x86_64/lib/clang-runtimes/arm-none-eabi/armv7m_hard_fpv4_sp_d16/include -IC:/Users/xxx/source/repos/TimerTest303RE/TimerTest_Nucleo_F303-RE/xxx/stm32cubemx/../../Core/Inc -IC:/Users/xxx/source/repos/TimerTest303RE/TimerTest_Nucleo_F303-RE/cmake/stm32cubemx/../../Drivers/STM32F3xx_HAL_Driver/Inc -IC:/Users/xxx/source/repos/TimerTest303RE/TimerTest_Nucleo_F303-RE/cmake/stm32cubemx/../../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -IC:/Users/xxx/source/repos/TimerTest303RE/TimerTest_Nucleo_F303-RE/cmake/stm32cubemx/../../Drivers/CMSIS/Device/ST/STM32F3xx/Include -IC:/Users/xxx/source/repos/TimerTest303RE/TimerTest_Nucleo_F303-RE/cmake/stm32cubemx/../../Drivers/CMSIS/Include --target=thumbv7m-unknown-arm-eabi -mfpu=fpv4-sp-d16 -mfloat-abi=hard --sysroot=F:/test/LLVM-ET-Arm-19.1.1-Windows-x86_64/lib/clang-runtimes/arm-none-eabi/armv7m_hard_fpv4_sp_d16 -fno-exceptions -fno-rtti -O0 -g3 -g -std=gnu11 -MD -MT CMakeFiles/TimerTest_Nucleo_F303-RE.dir/Core/Src/sysmem.c.obj -MF CMakeFiles\TimerTest_Nucleo_F303-RE.dir\Core\Src\sysmem.c.obj.d -o CMakeFiles/TimerTest_Nucleo_F303-RE.dir/Core/Src/sysmem.c.obj -c C:/Users/xxx/source/repos/TimerTest303RE/TimerTest_Nucleo_F303-RE/Core/Src/sysmem.c

> C:\Users\xxx\source\repos\TimerTest303RE\TimerTest_Nucleo_F303-RE\Core\Src\sysmem.c(30,35): error G13DA7946: use of undeclared identifier 'NULL'
> `     30 | static uint8_t *__sbrk_heap_end = NULL;

C:\Users\xxx\source\repos\TimerTest303RE\TimerTest_Nucleo_F303-RE\Core\Src\sysmem.c(53,13): error GC871EEFB: unknown type name 'ptrdiff_t'
  
     53 | void *_sbrk(ptrdiff_t incr)
  
        |             ^
  
C:\Users\xxx\source\repos\TimerTest303RE\TimerTest_Nucleo_F303-RE\Core\Src\sysmem.c(63,7): error G13DA7946: use of undeclared identifier 'NULL'
  
     63 |   if (NULL == __sbrk_heap_end)

 I compile it as C Code.
NULL is normaly declared in the stddef.h and i can also include it in the main.c.
Why it's fail to compile with clang and not GCC????

Bests
Pantau

voltur01, 2024-11-15:
FYI the doc for the newlib package https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/docs/newlib.md

voltur01, 2025-01-07:
Is this still relevant or is it OK for me to close?

Pantau123, 2025-01-07:
It's solved.. it can be closed

Mit freundlichen Grüssen
Brian Dahl
eMail: ***@***.***

Informdata Dahl
Brian Dahl
Bandwiesstrasse 3

8630 Rüti ZH, Schweiz

voltur01, 2025-01-07:
Thank you for the prompt confirmation!