ARM-software/LLVM-embedded-toolchain-for-Arm #631: For armv8.1m - Do we have hardfp+mve support?

amartsan, 2025-01-27:
I notice from the list of multlib json files available, we don't have a combination where MVE and hardfp supported.
https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/arm-multilib/json/variants/armv8.1m.main_hard_fp_nomve.json

Can you please clarify if this is possible or in plan for future?

smithp35, 2025-01-27:
There was a similar question in https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues/446 asking what the differences were between the libraries.

Probably best to read the answer there for the full context. A quick summary:
* The MVE and Hardfp is supported by the toolchain. While we don't have a library variant precompiled, we are able to map that configuration to armv8.1m.main_hard_fpdp_nomve family of library variabts which is compatible.
* The C-library and Math library, being scalar code, don't benefit significantly from MVE when there is already a floating point unit present. There are a tiny number of functions that see differences when compiled with floating-point and MVE support. As each new variant adds build and testing time we choose not to build it.

amartsan, 2025-01-29:
Thanks for the clarification.
As such code compiled from source generates MVE instructions and can work. This is only about C-runtime library options. So, this should not be an issue.

closed 2025-01-29 by amartsan