ARM-software/LLVM-embedded-toolchain-for-Arm #640: Could you clarify what changes have been made to LLVM 19?

SeanMeng0509, 2025-03-01:
Following up #446 

Upgrading from LLVM 18 to LLVM 19.1.x, noticed the size of libc.a (armv8.1m.main_hard_fpdp_nomve) was increased from 4419688 to 5054450. May we know what changes were made in LLVM 19? Any new features/functions were added?

voltur01, 2025-03-03:
Hi,

It is most likely related to this change https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/pull/441 making libraries for newer architectures build in release mode to improve performance. I do not believe there were any significant additions of functionality.

voltur01, 2025-03-03:
Yes, `malloc` was pointed out as a major reason for bad performance, discussed here https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues/383

newlib `malloc` is faster one included into release builds. It is decided here in the CMake file https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/f6f879ed45ff5727d821f6cd29571ee2baf31779/arm-runtimes/CMakeLists.txt#L351

voltur01, 2025-03-03:
Yes, it was a question in the process of review, however the change kept only smaller M-profile targets at `minsize`.

Unfortunately, there is no way to use `minsize` without rebuilding now. In the future, we want to see if we can create a model where we have both versions of `malloc` available and provide a command line option to chose one that is appropriate for the application.

voltur01, 2025-03-03:
`malloc` has a special option, changes in other functions, I assume, because of the `release` build in 19 vs `minsize` in 18 release.

closed 2025-03-03 by SeanMeng0509