ARM-software/LLVM-embedded-toolchain-for-Arm #641: Remaster picolibc patches

dobryj, 2025-04-08:
Hi,

While following the steps in building-from-source.md, I’ve run into an issue: the current picolibc repository state is no longer compatible with the patches applied during the build process. The build fails unless one reverts to an older version of picolibc, but the instructions do not specify which commit is known to work.

Would it be possible to either:

Update the patchset to be compatible with the latest picolibc,
or

Specify the exact picolibc commit that is expected to work, ideally right after the command:

git -C repos clone https://github.com/picolibc/picolibc.git

This would help avoid confusion and make the build process more reproducible.

Thanks a lot for your work!

voltur01, 2025-04-08:
Hi,

Thank you for your interest and the report!

The version of picolibc is fixed here https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/f6f879ed45ff5727d821f6cd29571ee2baf31779/versions.json#L16 so that CI build works out-of-box.

The toolchain was migrated to the new repository https://github.com/arm/arm-toolchain/tree/arm-software/arm-software/embedded and is being maintained there including the updated patches. So unless you have specific reasons to keep using this project, it would be better to move to the new one.

dobryj, 2025-04-08:
Hi, I saw the warning at the root of the referenced repo saying the project is under construction. 
I'm asking because we need to patch & recompile it because the standard distros still don't include a big-endian multilib variant for Cortex-R5 (armebv7r), and I've honestly been waiting for something like that for over 10 years now.

voltur01, 2025-04-09:
Hi,

We just published the first binary release from the new repo https://github.com/arm/arm-toolchain/releases/tag/release-20.1.0-ATfE and will remove the warning as it is a good enough assurance that the new scripts work as expected.