ARM-software/LLVM-embedded-toolchain-for-Arm #643: [rfc] Build picolibc with __PREFER_SIZE_OVER_SPEED

Damian-Nordic, 2025-05-19:
Picolibc provides size-optimized variants of certain standard functions, enabled with `__PREFER_SIZE_OVER_SPEED` or `__OPTIMIZE_SIZE__` macro. For example, https://github.com/picolibc/picolibc/blob/560946f26db075c296beea5b39d99e6de43c9010/newlib/libc/string/strstr.c#L79.

I think, since the toolchain is meant for embedded ARM, it would make sense to enable either macro to achieve lower footprint.

voltur01, 2025-05-19:
Hi,

Thank you for reaching out!

Which library variants are you interested in?

We have some variants build with code size optimization ("minsize" mode), e.g. Armv6, see this like in the JSON https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/c9dff0ac5ab961cf393b6189f27cadbde77dc02e/arm-multilib/json/variants/armv6m_soft_nofp.json#L21

but some other, e.g. Armv8.1-M, for performance ("release" mode), see this line https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/c9dff0ac5ab961cf393b6189f27cadbde77dc02e/arm-multilib/json/variants/armv8.1m.main_hard_nofp_mve.json#L21

We are thinking about a way to optimize layout of the libraries to minimize duplication of files and make it possible to provide both size and speed optimized versions of some key functions like memcopy, but it is still early discussions.

voltur01, 2025-05-19:
BTW, we moved development of the toolchain to the new project https://github.com/arm/arm-toolchain/tree/arm-software/arm-software/embedded so future changes will only be implemented there.

Damian-Nordic, 2025-05-19:
@voltur01 Thanks for the insights. I have been comparing GCC vs LLVM toolchain in terms of the generated code size for ARM cortex M-33 (ARMv8) and I noticed much larger string functions being included in the LLVM-generated binary. I was curious if this was an oversight but apparently this was a deliberate decision :).

Yeah, it would be great to have both size and speed optimized versions available so I'll look forward for any changes with regards to that!