ARM-software/LLVM-embedded-toolchain-for-Arm #644: [err] error in backend: Cannot select: intrinsic %llvm.arm.mve.max.predicated

bshaokang, 2025-06-17:
fatal error: error in backend: Cannot select: intrinsic %llvm.arm.mve.max.predicated

Stack dump:
0.      Program arguments: clang -c -Wstrict-prototypes -Wno-attributes -Wno-unknown-pragmas -Oz -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -fshort-enums -fno-common -Wall -Wshadow -Wundef -ffunction-sections -fdata-sections -target armv8.1m.main_hard_fp.cfg -march=armv8.1-m.main+mve.fp+fp.dp -mtune=cortex-m85 -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -Wa,-mthumb -Wa,-mimplicit-it=always -fstack-protector-strong

1.      <eof> parser at end of file
2.      Code generation
3.      Running pass 'Function Pass Manager' on module '/home/baishaokang/f3_kasan_6_17/vendor/xring/mcu_common/apps/vendor/contexthub/xr_turbo/nn/source/xrt_activation.c'.
4.      Running pass 'ARM Instruction Selection' on function '@xrt_relu_vec_f32'
 #0 0x0000000001efc018 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x1efc018)
 #1 0x0000000001efa294 llvm::sys::CleanupOnSignal(unsigned long) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x1efa294)
 #2 0x0000000001e53d83 llvm::CrashRecoveryContext::HandleExit(int) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x1e53d83)
 #3 0x0000000001ef3c9e llvm::sys::Process::Exit(int, bool) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x1ef3c9e)
 #4 0x0000000000ac64ed (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0xac64ed)
 #5 0x0000000001e5b775 llvm::report_fatal_error(llvm::Twine const&, bool) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x1e5b775)
 #6 0x0000000002eefaa1 llvm::SelectionDAGISel::CannotYetSelect(llvm::SDNode*) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x2eefaa1)
 #7 0x0000000002ef3882 llvm::SelectionDAGISel::SelectCodeCommon(llvm::SDNode*, unsigned char const*, unsigned int) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x2ef3882)
 #8 0x0000000000f26cc7 (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0xf26cc7)
 #9 0x0000000002eecb16 llvm::SelectionDAGISel::DoInstructionSelection() (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x2eecb16)
#10 0x0000000002efa51c llvm::SelectionDAGISel::CodeGenAndEmitDAG() (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x2efa51c)
#11 0x0000000002efce76 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x2efce76)
#12 0x0000000002eff706 (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x2eff706)
#13 0x0000000000f2fd24 (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0xf2fd24)
#14 0x000000000147a000 (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x147a000)
#15 0x00000000019e2608 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x19e2608)
#16 0x00000000019e277c llvm::FPPassManager::runOnModule(llvm::Module&) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x19e277c)
#17 0x00000000019e30ee llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x19e30ee)
#18 0x0000000002140cf3 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::HeaderSearchOptions const&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::IntrusiveRefCntPtr<llvm::vfs::FileSystem>, std::unique_ptr<llvm::raw_pwrite_stream, std::default_delete<llvm::raw_pwrite_stream>>, clang::BackendConsumer*) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x2140cf3)
#19 0x00000000026c9b69 clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x26c9b69)
#20 0x0000000003f821a9 clang::ParseAST(clang::Sema&, bool, bool) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x3f821a9)
#21 0x000000000293b179 clang::FrontendAction::Execute() (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x293b179)
#22 0x00000000028be85b clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x28be85b)
#23 0x00000000029f6063 clang::ExecuteCompilerInvocation(clang::CompilerInstance*) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x29f6063)
#24 0x0000000000ac81ce cc1_main(llvm::ArrayRef<char const*>, char const*, void*) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0xac81ce)
#25 0x0000000000abfce8 _start (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0xabfce8)
#26 0x000000000270d999 (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x270d999)
#27 0x0000000001e53c83 llvm::CrashRecoveryContext::RunSafely(llvm::function_ref<void ()>) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x1e53c83)
#28 0x000000000270dcf4 (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x270dcf4)
#29 0x00000000026d99dd clang::driver::Compilation::ExecuteCommand(clang::driver::Command const&, clang::driver::Command const*&, bool) const (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x26d99dd)
#30 0x00000000026da2ad clang::driver::Compilation::ExecuteJobs(clang::driver::JobList const&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&, bool) const (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x26da2ad)
#31 0x00000000026e93ec clang::driver::Driver::ExecuteCompilation(clang::driver::Compilation&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0x26e93ec)
#32 0x0000000000ac4d48 clang_main(int, char**, llvm::ToolContext const&) (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0xac4d48)
#33 0x0000000000a06d34 main (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0xa06d34)
#34 0x00007f87a4026083 __libc_start_main /build/glibc-LcI20x/glibc-2.31/csu/../csu/libc-start.c:342:3
#35 0x0000000000abf347 _start (/home/baishaokang/f3_kasan_6_17/vendor/xring/open-source/clang_for_arm/clang18/bin/clang-18+0xabf347)

bshaokang, 2025-06-17:
Did any errors in the compiler options lead to this err

bshaokang, 2025-07-01:
Update on the issue.
I found that the problem was caused by the compilation option -o3.
Compiling cmsis dsp failed when using -O3.
Compiling cmsis dsp succeeded when not using -O3.
Is this a bug?

smithp35, 2025-07-01:
Yes this is a bug, as the compiler shouldn't crash. However reporting the bug here isn't the right place for a couple of reasons:
* The LLVM embedded toolchain for Arm has been superseded by the arm-toolchain and is no longer being updated. Even if the bug is fixed in upstream clang it won't be fixed in this project.
* The bug is in upstream clang, it isn't specific to the toolchain. We need to report the bug to https://github.com/llvm/llvm-project/issues 

To increase the chances of the bug getting looked at it will help to:
* Try the most up to date toolchain, for example Clang-20 in https://github.com/arm/arm-toolchain/releases/tag/release-20.1.0-ATfE . If the bug has already been fixed there isn't much point in reporting it as bug fixes are not back-ported.
* Find the smallest possible reproducer for the crash. You mention CMSIS at -O3. That is a start but it would help to reduce it further. For example find the source file that crash occurs on, and use the preprocessor to make a reproducer. CMSIS is open-source so you can quote it in the bug report. https://clang.llvm.org/docs/UsersManual.html#cmdoption-gen-reproducer . Chances are it can be cut down further by finding the function that provokes the crash and deleting everything else.